platformio/platformio-atom-ide #772: InstallationManager

LBTRONIC, 2017-07-27:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\User\.platformio\.cache-ide\tmp-5680XhlLuSJcyzGv\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\User\.platformio\.cache-ide\tmp-5680XhlLuSJcyzGv\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\User\.platformio\.cache-ide\tmp-5680XhlLuSJcyzGv\virtualenv-14.0.6\virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "C:\Users\User\.platformio\.cache-ide\tmp-5680XhlLuSJcyzGv\virtualenv-14.0.6\virtualenv.py", line 1101, in install_python
    copy_required_modules(home_dir, symlink)
  File "C:\Users\User\.platformio\.cache-ide\tmp-5680XhlLuSJcyzGv\virtualenv-14.0.6\virtualenv.py", line 1041, in copy_required_modules
    dst_filename = change_prefix(filename, dst_prefix)
  File "C:\Users\User\.platformio\.cache-ide\tmp-5680XhlLuSJcyzGv\virtualenv-14.0.6\virtualenv.py", line 1015, in change_prefix
    (filename, prefixes)
AssertionError: Filename C:\python27\Lib\os.py does not start with any of these prefixes: ['C:\\Python27']

ivankravets, 2017-07-27:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
python --version
pip --version
pip list
```

LBTRONIC, 2017-07-27:
python version is 2.7.13
pip version is 9.0.1
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\User>echo %PATH%
C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\Wind
owsPowerShell\v1.0\;C:\PROGRA~1\PICC;C:\Program Files\Microchip\MPLAB C32 Suite\
bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\VISA
\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\Nati
onal Instruments\Motion Assistant\bin\mxwplugins;C:\Program Files\Microsoft SQL
Server\80\Tools\Binn\;C:\Program Files\Microsoft SQL Server\100\Tools\Binn\;C:\P
rogram Files\Microsoft SQL Server\100\DTS\Binn\;C:\Program Files\National Instru
ments\Shared\System\;C:\Program Files\PuTTY\;C:\Program Files\LLVM\bin;C:\Python
27;C:\Python27\Scripts\

C:\Users\User>PYTHON --VERSION
Unknown option: --
usage: PYTHON [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.

C:\Users\User>python --version
Python 2.7.13

C:\Users\User>pip --version
pip 9.0.1 from C:\Python27\lib\site-packages\pip-9.0.1-py2.7.egg (python 2.7)

C:\Users\User>pip list
DEPRECATION: The default format will switch to columns in the future. You can us
e --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.con
f under the [list] section) to disable this warning.
adafruit-ampy (1.0.1)
ampy (1.2.6)
click (6.7)
curses-2048 (1.4)
dateutils (0.6.6)
esptool (1.1)
Flask (0.12.1)
fonttools (2.3)
funcsigs (1.0.2)
gTTS (1.2.0)
gTTS-token (1.1.1)
itsdangerous (0.24)
Jinja2 (2.9.6)
MarkupSafe (1.0)
matplotlib (1.3.0)
MinimalModbus (0.7)
mock (2.0.0)
numpy (1.8.0)
pbr (1.10.0)
pip (9.0.1)
Polygon2 (2.0.6)
PyOpenGL (3.1.0)
pyparsing (2.1.10)
pypiwin32 (219)
pyserial (3.2.1)
python-dateutil (2.5.3)
pyttsx (1.1)
pytz (2016.7)
requests (2.13.0)
setuptools (28.8.0)
six (1.10.0)
speech (0.5.2)
UniCurses (1.2)
virtualenv (15.1.0)
Werkzeug (0.12.1)

C:\Users\User>

ivankravets, 2017-07-27:
Remobing of broken `PYTHONPATH` environment variable resolved issue.

closed 2017-07-27 by ivankravets