platformio/platformio-atom-ide #811: C/C++ project index rebuild failed

BBorowy, 2017-08-05:
Exception: [Errno 13] Permission denied: '/Users/BogdanBorowy/Documents/MyProjects/TestESP12/.pioenvs/do-not-modify-files-here.url':
File "/Users/BogdanBorowy/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 93:

PROJECTPIOENVS_DIR=util.get_projectpioenvs_dir(),
File "/Users/BogdanBorowy/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 309:
raise Exception(e)

========================== [ERROR] Took 2.27 seconds ==========================

nk36, 2017-08-24:
Hi,

I'm having the same problem since 10 minutes and I don't know why..

Have you found a solution ??

Thanks

nk36, 2017-08-24:
Ok, I found the solution by myself.. I just remove the .pioenvs in my project!

ivankravets, 2017-09-23:
Please remove `/Users/BogdanBorowy/Documents/MyProjects/TestESP12/.pioenvs`

P.S: Fixed issue in the latest PIO Core

closed 2017-09-23 by ivankravets