platformio/platformio-atom-ide #812: InstallationManager

Blackhex, 2017-08-05:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\black\.platformio\.cache-ide\tmp-176529FXb3ErmRBRo\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\black\.platformio\.cache-ide\tmp-176529FXb3ErmRBRo\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\black\.platformio\.cache-ide\tmp-176529FXb3ErmRBRo\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\black\.platformio\.cache-ide\tmp-176529FXb3ErmRBRo\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\black\.platformio\.cache-ide\tmp-176529FXb3ErmRBRo\virtualenv-14.0.6\virtualenv.py", line 747, in call_subprocess
    "Error %s while executing command %s" % (e, cmd_desc))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xed in position 18: ordinal not in range(128)

ivankravets, 2017-10-17:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

Blackhex, 2017-10-17:
Here it is:

`
echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\Python27\;C:\Python27\Scripts;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files (x86)\Microsoft SQL Server\110\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\130\DTS\Binn\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\nodejs\;C:\Program Files\LLVM\bin;C:\Program Files (x86)\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files\PuTTY\;C:\Users\black\AppData\Local\Microsoft\WindowsApps;C:\Users\black\AppData\Roaming\npm;C:\Users\black\AppData\Local\atom\bin;C:\Program Files\Microsoft VS Code\bin

echo %PYTHONPATH%
%PYTHONPATH%

python --version
Python 2.7.13

pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
arrow (0.10.0)
bottle (0.12.13)
certifi (2017.7.27.1)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
idna (2.6)
lockfile (0.12.2)
lxml (3.7.3)
pip (9.0.1)
platformio (3.5.0a9)
pyserial (3.4)
python-dateutil (2.6.1)
requests (2.18.4)
semantic-version (2.6.0)
setuptools (28.8.0)
six (1.10.0)
urllib3 (1.22)
`

ivankravets, 2017-10-18:
Please run in Terminal (cmd) this command
```
pip install virtualenv
```

Restart IDE. Does it work now?

Blackhex, 2017-10-18:
Honestly, I am not sure what was wrong when I was trying PlatformIO and even how the issue was submitted to you. It was not submitted explicitly by me but somehow automatically. I am even not sure if it was a IDE crash or some other malfunction. But I can confirm that now, that after virtualenv installation, I can import and build Marlin 3D printer firmware which, I think, I could not before. I was trying many different IDEs for Android projects compilation that day so I am sorry I cannot provide you more detailed information what happened back then.

closed 2017-10-18 by ivankravets