platformio/platformio-atom-ide #824: InstallationManager

onpagithub, 2017-08-10:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\e202120\.platformio\.cache-ide\tmp-17104UGHI87T3mAdV\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\e202120\.platformio\.cache-ide\tmp-17104UGHI87T3mAdV\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\e202120\.platformio\.cache-ide\tmp-17104UGHI87T3mAdV\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\e202120\.platformio\.cache-ide\tmp-17104UGHI87T3mAdV\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\e202120\.platformio\.cache-ide\tmp-17104UGHI87T3mAdV\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\e202120\.pl...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-10-17:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/703

closed 2017-10-17 by ivankravets

ivankravets, 2017-10-17:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

What is your Windows OS version? Locale?