platformio/platformio-atom-ide #833: Distribution number mismatch? +Failed to get boards

hexeguitar, 2017-08-12:
After the last core update to 3.5.0a4 i can't init a new board, getting the 
`Failed to get boards error`
similar to #484 
What i also noticed, opening up the terminal shows an error at start, the "pio --help" command throws up an error:
`pkg_resources.DistributionNotFound: The 'platformio==3.5.0a3' distribution was not found and is required by the application`
while i have the latest 3.5.0.a4 installed (tried reinstalling it too - no avail).
Here is the screenshot:
http://i.imgur.com/9yZ1HNs.png

On the other hand, using command "platformio" instead of "pio" works fine, looks like "pio" is pointing to a previous version of the core.

Thanks!
P.

hexeguitar, 2017-08-16:
Not sure why did it happen, perhaps the update process has not been finished properly, here is a fix for it on Win764bit:
go to
`C:\Users\USERNAME\.platformio\penv\Scripts\pio-script.py`
and change the version number 3.5.0a3 to 3.5.0a4 in line 3 and 11.

amayii0, 2017-09-05:
Also got this error but about platformio==3.5.0a8
Also after an update of pio, nothing fancy done, plain regular self update

pio-script.py contains 3.5.0a8 while platformio upgrade reports 3.5.0a9
pretty annoying, it looks like dependencies often break during update process :( had issues in the past with minimap, terminal and so on

ivankravets, 2017-09-05:
Please remove `C:\Users\USERNAME\.platformio\penv` folder and restart Atom

closed 2017-09-05 by ivankravets