platformio/platformio-atom-ide #844: C/C++ project index rebuild failed

ericklil, 2017-08-15:
Error: Traceback (most recent call last):
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 104, in main
    cli(None, None, None)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 39, in __init__
    self._gather_tplvars()
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 152, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 134, in __call__
    value = self.func(*args)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 74, in get_project_build_data
    target=["idedata"])
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)

  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 109, in cli
    result = (envname, ep.process())
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 193, in process
    result = self._run()
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 287, in _run
    p = PlatformFactory.newPlatform(self.options['platform'])
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 212, in newPlatform
    name, requirements)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 308, in get_package_dir
    manifest = self.get_package(name, requirements, url)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 278, in get_package
    for manifest in self.get_installed():
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 268, in get_installed
    manifest = self.load_manifest(pkg_dir)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 232, in load_manifest
    src_manifest_path = self.get_src_manifest_path(pkg_dir)
  File "/Users/ericko/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 203, in get_src_manifest_path
    for item in os.listdir(pkg_dir):
OSError: [Errno 13] Permission denied: '/Users/ericko/.platformio/platforms/espressif8266'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ericklil, 2017-08-15:
tried all that I could find, to no avail, what did I overlook, Grateful for any pointers

ivankravets, 2017-10-17:
Please remove `/Users/ericko/.platformio` adn restart IDE

closed 2017-10-17 by ivankravets