platformio/platformio-atom-ide #849: InstallationManager

HaTWare, 2017-08-17:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\I046746\.platformio\.cache-ide\tmp-115760pQGJB29AdAh\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\I046746\.platformio\.cache-ide\tmp-115760pQGJB29AdAh\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\I046746\.platformio\.cache-ide\tmp-115760pQGJB29AdAh\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\I046746\.platformio\.cache-ide\tmp-115760pQGJB29AdAh\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\I046746\.platformio\.cache-ide\tmp-115760pQGJB29AdAh\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\I046746\.pl...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-10-17:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/703

closed 2017-10-17 by ivankravets

ivankravets, 2017-10-17:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

What is your Windows OS version? Locale?