platformio/platformio-atom-ide #869: platform-ide package installation in Atom fails

LeonhardScheck, 2017-08-23:
I tried to install the platformio-ide package (in Atom 1.19.3 on MacOS 10.12.6) and get the following error:

> "Failed to activate the platformio-ide package
> Unexpected token import
> 
> The error was thrown from the platformio-ide package. This issue has already been reported."

And there is a link to [Issue 752](https://github.com/platformio/platformio-atom-ide/issues/752) , which is closed (and looks different to me)... 

Stack Trace:

```
SyntaxError: Unexpected token import
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:94:52)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/node_modules/react-router/MemoryRouter.js:13:28)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/node_modules/react-router/MemoryRouter.js:62:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/node_modules/react-router/index.js:6:22)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/node_modules/react-router/index.js:53:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/account/containers/auth-container.js:27:20)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/account/containers/auth-container.js:28:35)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/account/sagas.js:15:18)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/account/sagas.js:265:2)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/core/sagas.js:15:65)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/core/sagas.js:41:2)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/core/store.js:15:37)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/core/store.js:72:2)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/main.js:17:64)
    at Object.<anonymous> (/Users/leonhardscheck/.atom/packages/platformio-ide/lib/main.js:279:22)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:96:26)
    at Package.module.exports.Package.requireMainModule (/Applications/Atom.app/Contents/Resources/app/src/package.js:887:35)
    at Package.module.exports.Package.activateNow (/Applications/Atom.app/Contents/Resources/app/src/package.js:242:22)
    at /Applications/Atom.app/Contents/Resources/app/src/package.js:1014:34
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at /Applications/Atom.app/Contents/Resources/app/src/package-manager.js:658:56
    at Set.forEach (native)
    at PackageManager.module.exports.PackageManager.activatePackage (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:656:47)
    at /Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:535:35
    at exit (/Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:128:22)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:338:17)
    at ChildProcess.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:368:17)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

neu-rah, 2017-08-23:
me too, error after atom-update and does not go away by uninstall/reinstall platformio

neu-rah, 2017-08-23:
I'm using multiple profiles for atom as I use it for many heavy and diverse environments...
making the install on the **default** profile seem to be ok

the env selector is just a script named atom that accepts a parameter and sets 

   ATOM_HOME="$atom_root/$profile"

however it does so for **default**.. fact is that it solved for me

LeonhardScheck, 2017-08-23:
I did not do anything to set up or change profiles, so probably I am using the default profile.
I uninstalled Atom, deleted ~/.atom, reinstalled Atom and still get the same error when I try to install the platformio-ide package...

ivankravets, 2017-10-17:
Have you resolved this issue?

closed 2017-11-06 by ivankravets