platformio/platformio-atom-ide #885: Arduino.h: file not found in the editor, but not in the compiler

samerafach, 2017-08-29:
In Visual Studio Code, the part  `#include "Arduino.h"` is underlined with red. This happens all the time, although the compiler works fine and everything compiles and uploads to the device. Here's [a screenshot](http://imgur.com/a/icuQJ).

Now if I change `#include "Arduino.h"` to something totally invalid, say `#include "Arduino1.h"`, the error message is different. Here's [another screenshot](http://imgur.com/a/mLISf).

Is this a bug or a misconfiguration from my end?

ivankravets, 2017-09-06:
Please wait while VSCode rebuilds index...

samerafach, 2017-09-10:
Thanks for your response. I'm sorry, I don't understand what that means. The program has been running for days, and there's nothing loading. Is there a way to actively activate this? Your comment made me look for this, and I ran "Rebuild Intellisense Index" command in VS Code, and got the result "PlatformIO: IntelliSense Index has been successfully rebuilt.", but that didn't change the red underline status and the problem still persists. Could you please elaborate or refer me to the docs that discuss this issue?

ivankravets, 2017-09-11:
Please provide your `platformio.ini` file.

samerafach, 2017-09-11:
OK, after rebuilding the index, and actively changing the text forth and back from `Arduino.h`, it was able to find it. Thank you for your time :)

closed 2017-09-11 by samerafach

almirb, 2018-07-10:
For me, that was happening after installing the mitaki28.vscode-clang extension.
After disabled It, the red underline disappeared.

Jrexexe, 2018-12-17:
disabled mitaki28.vscode-clang extension than ok

StipeElektrotechnik, 2019-12-28:
How i make the index for Arduino uno?