platformio/platformio-atom-ide #905: warning: Two different environment were specified for target ...

danieldbb, 2017-09-05:
At the start of build process : reports (see image)
... I cannot copy paste text at mac platformio ide, selecting selects all text in every window

[image]

here is my platformio.ini

[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = espidf
build_flags =
lib_extra_dirs =
  lib/components/deps/
  lib/components/ili9341/

second building pass:

[image]

and at the end:
[image]

Thanks for help!
Kind regards,
Daniel

ivankravets, 2017-09-05:
Could you provide test project to reproduce this issue?

danieldbb, 2017-09-06:
Hello Igor,

first of all - gratulation for great product!

Perhaps I did something wrong ... i just follow guiding instruction

zip file   .... cleaned Project

2nd question ...

is this a bug?
- when I would llke to select words (of text) in Build Panel all text is
going to be selected,
- also if I would like to jump to the line from Build Panel it wont work.

That's all for this moment.

Kind regards,
Daniel

ivankravets, 2017-09-06:
Thanks for the report!

Please switch to PlatformIO IDE for VSCode for better experience and performance:

- http://docs.platformio.org/en/latest/ide/vscode.html

danieldbb, 2017-09-11:
Hello Ivan,

strange think hapend today ...
in Platformio It's not possible to create new project.
Yesterday a could create one today is not possible.

Yesterday I could create new Project in default dir and during the process
of creation Platformio asserts new folder and did the rest of files and
objects.
Today, I cannot Add an empty project folder, not even the name.

The failure I wrote you one day about  "Two different environment..." now
happends in VSCde to.
I cannot compile programms thatI could compile yesterday.

Did you have new version?

KInd regards,
Daniel

ivankravets, 2017-09-12:
Please share your `platformio.ini` file from a project

ivankravets, 2017-10-17:
Please reopen if you still need help

closed 2017-10-17 by ivankravets

YvedD, 2017-10-20:
Hi,

I had some problems before, so i re-installed the VSCode + Platformio IDE.
Then i did "pio lib update" & "pio lib upgrade"
Now i keep getting this error on "build" : 

*** Two environments with different actions were specified for the same target: C:\Users\Beheerder\Documents\PlatformIO\Projects\b
le_hello_world\.pioenvs\nodemcu-32s\app_trace\app_trace.o
(action 1: xtensa-esp32-elf-gcc -o app_trace.o -c -Wno-old-style-declaration -std=gnu99 -Wall -Werror=all -Wno-error=deprecated-de
clarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=unused-function -Og -g3 -nostdlib -Wpointer-arith -Wno-error
=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields $(
subst --coverage,,$(CFLAGS)) -DWITH_POSIX -DIDF_VER=\"2.210.0\" -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTL
S_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPLATFORMIO=30500 -DESP32_DEV -DARDUINO_NodeMCU-32S @"c:\Users\Beheerder\Documents\Platfor
mIO\Projects\ble_hello_world\.pioenvs\nodemcu-32s\longcmd-1e799d0546b2a6908cf29a0079afec45" app_trace.c)
(action 2: xtensa-esp32-elf-gcc -o app_trace.o -c -Wno-old-style-declaration -std=gnu99 -Wall -Werror=all -Wno-error=deprecated-de
clarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=unused-function -Og -g3 -nostdlib -Wpointer-arith -Wno-error
=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -D
WITH_POSIX -DIDF_VER=\"2.210.0\" -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_con
fig.h\" -DPLATFORMIO=30500 -DESP32_DEV -DARDUINO_NodeMCU-32S @"c:\Users\Beheerder\Documents\PlatformIO\Projects\ble_hello_world\.p
ioenvs\nodemcu-32s\longcmd-1e799d0546b2a6908cf29a0079afec45" app_trace.c)
File "C:\Users\Beheerder\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 272, in BuildLibrary

I really don't know what i did wrong or how to fix this
Grtz, Yves

YvedD, 2017-10-20:
My platformio.ini : 
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = espidf
monitor_baud = 115200

telamon, 2018-08-17:
I have the same issue but using the commandline: 

     $ mkdir issue/ && cd issue/ 
     $ platformio init --board uno
     $ platformio run
```
[Fri Aug 17 16:27:49 2018] Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

*** Multiple ways to build the same target were specified for: /tmp/issue/.pioenvs/uno/FrameworkArduino/wiring_pulse.o  (from ['/home/telamon/.platformio/packages/framework-arduinoavr/cores/arduino/wiring_pulse.S'] and from ['/home/telamon/.platformio/packages/framework-arduinoavr/cores/arduino/wiring_pulse.c'])
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 273, in BuildLibrary
========================== [ERROR] Took 0.36 seconds ==========================
``` 

What to do?

ivankravets, 2018-08-17:
@telamon it seems that you use outdated PlatformIO Core and dev/platforms. Please open PlatformIO IDE Terminal and type:

```
$ pio upgrade
$ pio update
$ pio run
```

telamon, 2018-08-20:
@ivankravets thanks, that did the trick.