platformio/platformio-atom-ide #906: Seems stuck while updating/installing.

cranphin, 2017-09-05:
Today when I started Atom, I got the 'PlatformIO IDE: Installing...' dialog, which seems to be stuck on the PlatformIO Code item, there's a icon with a screen and a down arrow blinking next to it.

I've tried deleting everything from Atom and PlatformIO, reinstalling, and I get stuck in almost the same way, only now it's at 'Atom dependencies' and 'PlatformIO Core', both blink :)

Closing and restarting doesn't help, except that 'Atom dependencies no longer blinks'.

I've waited rather a long time, so it looks like it's not still busy. Shouldn't take too long on a SSD and fast PC anyways. I see 0 network traffic also.

Is there any way to debug this? Any log files?

bgulla, 2017-09-05:
I am having the same issue when installing PlatformIDE on osx from scratch.

madscientist42, 2017-09-05:
I honestly WISH they'd quit arbitrarily updating this willy-nilly.  I like that it provides a clean IDE- but leaving at least some of us DEAD IN THE WATER is not acceptable.  You want paying customers?  You _NEED_ to quit doing this and let us have some say in updates, etc. along with some ability to UN-break it by grabbing a previous version.

bgulla, 2017-09-05:
Does anyone know if there is a way on a mac to open atom and see the logging output of what is actually breaking the install?

madscientist42, 2017-09-05:
I'm trying to figure it out on Windows right now.  Not a lot of feedback- worse, this should be yet another APM package and it's technically not.  Tells me they're hiding things.  I'll update the issue if I find out a way to get Atom to tell us what the failure is within the rest.  I've real issues with what they've done here and how they've done it.

madscientist42, 2017-09-05:
"Error: spawn virtualenv ENOENT", source: C:\Users\Fearl\.atom\packages\platformi
o-ide\node_modules\platformio-node-helpers\lib\index.js (2395)"

Can't say that this is the culprit, though.  It's just suspicious.

How you get anything is from a command console, issue the command, "atom -f" which forces it to stay in the foreground of the console.  This works from Windows, Linux, or OSX.

ivankravets, 2017-09-05:
`Menu > View > Developer > Toggle Developer Tools `

Guys, please provide your console outputs. Thanks!

bgulla, 2017-09-05:
@madscientist42 looks like im seeing the same thing you are. 

@ivankravets thanks for the tip, here's my log. 
```

/Users/bgulla/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2395 User's Virtualenv: Error: spawn virtualenv ENOENT
_callee8$ @ /Users/bgulla/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2395
/Users/bgulla/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:458 

runCommand Array[5] function (code, stdout) {
                if (code !== 0) {
                  return reject(null);
                }
                resolve(stdout.trim() === 'null' ? null : stdout.trim());
         … Object
```

ivankravets, 2017-09-05:
Could you remove `%HOME_DIR/.platformio/penv` folder and restart Atom?

madscientist42, 2017-09-05:
Actually determined what MY problem actually was.  There *IS* a bug, but it's self-inflicted as to the trigger.

If, for any reason, it thinks you've got a native, already installed Python 2.7, the scripting appears to want to use it.  If it can't find one, it helpfully tries to add it.  One of the things the scripting currently does is set up a Python virtualenv.  Without having your path set so it can find the pip added stuff (including pip, through get_pip.py or similar...) it just barfs out trying to set up the virtualenv and never completes, leaving the "helpful" UI notifications from the upgrade manager application.

I'd recently corrected a bunch of path errors and inconsistencies (like on Friday) before leaving for the day.  Platformio could NOT find virtualenv as a command (as indicated above in one of my prior posts.)- as soon as I sorted out (and installed virtualenv in THAT Python setup, everything started working again.

Check your path.  Make sure you can issue a "pip" command and a "virtualenv" from the command line of your system.  I can't say for sure if this is a generic fix for everyone or not, but it corrected my issue and I'm back "up"

ivankravets, 2017-09-05:
@madscientist42 thanks a lot for your report. Please note that PIO Installer automatically downloads `virtualenv` and uses it to create the new virtual environment for PlatformIO Core. So, you don't need to install it before.

It seems that I understand the type of this problem. This issue could arise when user modifies Python Installation after PIO Installer created virtual environment. In this case, we have broken `%HOME_DIR%/.platformio/penv`. I'll think on that how to avoid similar issue.

madscientist42, 2017-09-05:
Cool.  It looked like that this was the case, but I wasn't for-sure because I only had my understanding of things...which comes from what I'm currently working with (which is the Client's Windows setup...  My mahcine?  LINUX...  X-D).

It's just frustrating beyond words when your toy (or tool in my case) breaks on you for no good reason.  Now that we know there's a sharp edge, we can watch out for it until you get the proper fix in the codebase.

ivankravets, 2017-09-05:
Indeed, it is very hard to maintain this extension for Atom and provide C/C++ instruments. Atom is not ready out the box for these tasks. We depend on different plugins, provide own extensions, own hooks to make things working.

We've started a work on [PlatformIO IDE for VSCode](https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide) a few months ago. Today, our integration with VSCode is MUCH BETTER than with Atom. VSCode provides rich API for plugin developers. We highly recommend switching to VSCode and our extension for the best performance.

P.S: We will maintain this extension for Atom but will recommend new users to get started from VSCode.

bgulla, 2017-09-05:
@ivankravets fyiw, I am on osx 10.11 and my ```~/.platformio``` only contains a ```.cache``` folder and no ```penv``` directory. Any more ideas to move past this? I have also wiped Atom from the box and reinstalled to no avail.

madscientist42, 2017-09-05:
You'll have to excuse me while I chuckle at the notion of using VS Code there.  In reality, you can't trust MS to not ream you in the end.  Seriously.

madscientist42, 2017-09-05:
It'd be better to find (or even MAKE) a better UI guys...  ;-D

ivankravets, 2017-09-05:
@madscientist42 VSCode is free, open-source, MIT license, and uses the same Electron engine like Atom.

madscientist42, 2017-09-05:
Source license is not of concern.  Patents are where MS gets you...  IF there's no patent license issues, then yeah, I'm all for it if it's cleaner than this is (which was NICE even with the issues...)- but are you sure there's no licensed stuff in there?  Better count fingers and toes...they really can't be trusted with 30+ years of screwing people after claiming they'd "changed".

Anyhow...you probably ought to make SURE you aren't at-risk from any patent claims and if it's as you claimed and it's not got patent landmines, you're right, you'd be stupid to not go there.

bgulla, 2017-09-05:
update: i was able to get things going after installed virtualenv (i guess my addiction to docker relieved my need for virtualenv on this box).

weird seeing how it's supposed to be downloaded with the install.

ivankravets, 2017-09-05:
VSCode was in a bad "position" 1.5 years ago when we started a work on PIO IDE for Atom. It had MS's proprietary license and boring API. That is why we decided to use Atom.

Today, they totally changed policy and left the community to decided how VSCode should be developed. They maintain it using own resources. 

We don't promote Github or Microsoft. Our aim is to provide best instruments in pair with PIO Core.

FYI: https://github.com/Microsoft/vscode/blob/master/LICENSE.txt

ivankravets, 2017-09-05:
Wow!!! Sorry!!! :((( That is really a bug on our side with "file downloader" :((( So, as result, installer could not download virtualenv :(

I'll fix it as soon as possible.

cranphin, 2017-09-05:
I will try to provide the named logs when I have some time again, in my case there is no .platformio/penv folder since I deleted .platformio and everything else trying to solve this :)
Keep in mind my case is on Windows (10).
There is a standard python 2.7.11 installed on my machine, tho I haven't changed anything in/to it, otherwise things should be pretty clean.

ivankravets, 2017-09-05:
@cranphin no need to provide any logs. I've just reproduced this issue in Virtual Machine. I'm working on a fix.

cranphin, 2017-09-05:
And yes, it's annoying spending two days on platformio issues instead of what I wanted to work on, but I'll take that over closed source any day ;)

robthepyro, 2017-09-05:
Also been struggling with this issue for the last 2 days, thought it was me because I was messing around with my Python install.. 
Ended up reinstalling Atom / Python / PIO multiple times. Tried installing PIO core with PIP first as well. (Win10x64)  

Eventually found this thread and installed virtualenv using pip from command line, can confirm PIO ide installed fine in Atom after that... 

Perhaps should have saved myself the trouble and tried PIO in VS Code haha. .

ivankravets, 2017-09-05:
The bad guy, Ivan.... We should punish him 🔥 

Fixed in https://github.com/platformio/platformio-node-helpers/commit/9b33897e422541b73707c22e772b1d69ddef32f7

So, new users will use updated package automatically. 

## IF YOU STUCK WITH INSTALLER POPUP, PLEASE REMOVE `%HOME_DIR%/.atom/packages/platformio-ide` FOLDER AND INSTALL PACKAGE AGAIN.

ivankravets, 2017-09-05:
Sorry, I reopen this issue, maybe, it will be useful for others.

MarcusWolschon, 2017-09-23:
Stuck with the installer popup, deleted packages/platformio-ide. started atom, installed platformio-ide again using file->settings->install, stuck at installer popup again.

closed 2017-10-17 by ivankravets

pea-arthur, 2017-12-17:
I've been encountering this bug on my chromebook (Gallium) multiple times over months. I tried reinstalling my OS, Atom, and everything over and over with the same result. I remember encounter the "updating dependencies" several times, hanging my system preventing me from coding. This is not fixed. Just yesterday I tried again and I the above advice to remove the folder. Now when I try to install the package, I get this output:

` pty.js@0.3.0 install /tmp/apm-install-dir-1171116-32304-m6li05.9m35vwvcxr/node_modules/platformio-ide-terminal/node_modules/pty.js
> node scripts/install.js

/tmp/apm-install-dir-1171116-32304-m6li05.9m35vwvcxr
└── (empty)

module.js:471
    throw err;
    ^

Error: Cannot find module '/tmp/apm-install-dir-1171116-32304-m6li05.9m35vwvcxr/node_modules/platformio-ide-terminal/node_modules/pty.js/scripts/install.js'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:509:3
npm WARN enoent ENOENT: no such file or directory, open '/tmp/apm-install-dir-1171116-32304-m6li05.9m35vwvcxr/package.json'
npm WARN apm-install-dir-1171116-32304-m6li05.9m35vwvcxr No description
npm WARN apm-install-dir-1171116-32304-m6li05.9m35vwvcxr No repository field.
npm WARN apm-install-dir-1171116-32304-m6li05.9m35vwvcxr No README data
npm WARN apm-install-dir-1171116-32304-m6li05.9m35vwvcxr No license field.
npm ERR! Linux 4.8.17-galliumos
npm ERR! argv "/opt/atom/resources/app/apm/bin/node" "/opt/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/peaarthur/.atom/.apm/.apmrc" "--userconfig" "/home/peaarthur/.atom/.apmrc" "install" "/tmp/d-1171116-32304-1js8qxn.kg0f1rlik9/package.tgz" "--runtime=electron" "--target=1.6.15" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! pty.js@0.3.0 install: `node scripts/install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the pty.js@0.3.0 install script 'node scripts/install.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the pty.js package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/install.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs pty.js
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls pty.js
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/apm-install-dir-1171116-32304-m6li05.9m35vwvcxr/npm-debug.log
npm ERR! code 1
`