platformio/platformio-atom-ide #909: "Seems like PlatformIO IDE is already being installed in another window." on startup

geekerator, 2017-09-05:
(Note: Re-occurrence of a closed thread - #113.)
I also get this message, but in another way. I:
  Started atom (with Platformio previously installed) and did an update to the latest core
  Platformio Core flashing download icon never stopped (after about 2 hrs) - and reappears on any restart
  No work can be done while Installing is going on, so I
  Quit and restarted while randomly clicking on atom's menu items, then
  This message comes up. I don't seem to have an install.lock file anywhere
My specs are: openSuSE 42.3 and kernel 4.4.79-19. Any suggestions would be greatly appreciated...

ivankravets, 2017-09-05:
Could you remove `~/.platformio/penv` folder and restart Atom?

geekerator, 2017-09-05:
Thanks for the quick reply, but ~/.platformio/penv folder does not exist on my machine. It almost seems like the download target is down.
This is what I have:
~/.platformio
    /.cache
    /.cache-ide
    /lib
    /packages
    /platforms
    /project-examples
    appstate.json

geekerator, 2017-09-05:
appstate.json contents are:
{"last_version": "3.4.1", "cid": "30cc3611-a7e5-e3a5-af5f-2e225bfaf9f2", "telemetry": {"backup": []}, "last_check": {"platformio_upgrade": 1504472174, "platforms_update": 1504552503, "libraries_update": 1504552505}}

ivankravets, 2017-09-05:
Fixed in https://github.com/platformio/platformio-atom-ide/issues/906

closed 2017-09-05 by ivankravets