platformio/platformio-atom-ide #914: 2.0.0-rc.1: loop after uncheck "Use build-in PlatformIO Core"

rwalli, 2017-09-06:
Also tried a fresh install on macOS 10.12.6 / Atom 1.19.5:
cd ~
rm -fr .atom
rm -fr .platformio
atom

install platformio-ide in atom
-> Restart
uncheck "Use build-in PlatformIO Core" in platformio settings

This message loops after each restart of Atom:
-
PlatformIO IDE has been successfully installed!
Please restart Atom to apply the changes
-

ivankravets, 2017-09-06:
So, you unchecked option "Use built-in PIO Core"... Do you have global PIO Core in a system?

rwalli, 2017-09-06:
Platformio-ide has installed a version of platformio which is usable from the command line.
On prior versions of platformio-ide it was possible to uncheck "Use build-in PlatformIO Core" and use this version.
If that is not possible any more platformio-ide should maybe produce a warning instead of the loop?

I want to use a common version of platformio for atom and vscode.
What is the way to do this?

ivankravets, 2017-09-06:
See description here https://github.com/platformio/platformio-atom-ide/blob/develop/lib/config.js#L53

So, if you uncheck built-in PIO Core, you should provide in System PATH own version of PIO Core. I don't recommend to uncheck this options. It was added for testing and plugin developers.

-----

Atom and VSCode use common built-in PIO Core which is located in `%HOME_DIR%/.platformio/penv`  folder. So, please check this option back 😊

closed 2017-09-06 by ivankravets