platformio/platformio-atom-ide #918: PlaformIO Home does not show

eczajk1, 2017-09-07:
I just did a fresh install of Atom and PlatformIO. The PlatformIO Home tab opens, but all I see is a spinning graphic:

[image]

I'm not an expert in debugging Atom, but I do see this message in the developer console:

```
internal/process/next_tick.js:103 Uncaught (in promise) Could not start PIO Home server: Error: timeout
```

I am on Windows 10:

```
atom --version

Atom    : 1.19.6
Electron: 1.6.9
Chrome  : 56.0.2924.87
Node    : 7.4.0
```

ivankravets, 2017-09-07:
Please open PlatformIO IDE Terminal and type
```
pio home
```

What do you see?

eczajk1, 2017-09-07:
When I try to type in the terminal, I get this issue: https://github.com/platformio/platformio-atom-ide-terminal/issues/350

mac-bleck, 2017-09-21:
I'm with the same problem, have you found the error?

mac-bleck, 2017-09-21:
When I type pio home this appears:

PS C:\Users\Maurcio\.atom> pio home
CorePackageManager: Installing tool-pioplus @ >=0.10.6,<2
Downloading...
Unpacking...
Warning! Package Mirror: [Error 5] Acesso negado: u'C:\\Users\\Maurcio\\.platformio\\packages\\tool-pioplus\\pioplus.exe'
Looking for other mirror...
Downloading...
Unpacking...
Warning! Package Mirror: [Error 5] Acesso negado: u'C:\\Users\\Maurcio\\.platformio\\packages\\tool-pioplus\\pioplus.exe'
Looking for other mirror...
Error: Could not install 'tool-pioplus' with version requirements '>=0.10.6,<2' for your system 'windows_x86'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

is not the antivirus, disabled and even so it continues giving this error

ivankravets, 2017-09-21:
Please remove `C:\Users\Maurcio\.platformio` folder and start again

japina, 2017-09-27:
The same happens to me. Tried to remove .platformio but it doesn't help.
pio home produces: ImportError: No module named platform

although the module can be found in shell (outside platformio) when running python.

ivankravets, 2017-10-17:
Have you resolved this issue?

japina, 2017-10-17:
Unfortunately no. I tried to delete .platformio few times, but nothing :(
Don't even know where to look to try to debug it.

erkobg, 2017-10-17:
After a fresh install - removed Atom, Platformio, folders.... It is resolved for me.

ivankravets, 2017-10-17:
@japina what is your OS?

japina, 2017-10-17:
Mac OSX.

ivankravets, 2017-10-17:
@japina please remove `~/.platformio/penv` folder and restart IDE

japina, 2017-10-17:
I remove the whole ~/.platformio many times, but it didn't help. Did remove ~/.platformio/penv right now, but still nothing :/

ivankravets, 2017-10-17:
@japina please open PlatformIO IDE Terminal and type
```
pio home
```

Do you see any errors?

japina, 2017-10-17:
The error that I mentioned previous:

Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 15, in init cmd_debug
ImportError: No module named platform

ivankravets, 2017-10-17:
@japina  Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

japina, 2017-10-17:
1. empty
2. empty
3. Python 2.7.10
4. pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)
5. DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in y
our pip.conf under the [list] section) to disable this warning.
arrow (0.10.0)
backports-abc (0.5)
bleep (0.4.4.dev3)
bottle (0.12.13)
certifi (2017.4.17)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
configobj (5.0.6)
Flask (0.12.2)
future (0.16.0)
glob2 (0.6)
idna (2.5)
ino (0.3.7)
itsdangerous (0.24)
Jinja2 (2.9.6)
lockfile (0.12.2)
MarkupSafe (1.0)
mercurial (4.3.2)
pip (9.0.1)
PyBluez (0.22)
pycairo (1.15.3)
pycam (0.6.1)
pygattosx (0.0.1)
pygobject (3.26.0)
pyobjc-core (4.0)
pyobjc-framework-Cocoa (4.0)
pyportmidi (0.0.7)
pyserial (3.2.1)
python-dateutil (2.6.0)
requests (2.18.1)
semantic-version (2.6.0)
setuptools (36.5.0)
singledispatch (3.4.0.3)
six (1.10.0)
tornado (4.5.2)
urllib3 (1.21.1)
virtualenv (15.1.0)
Werkzeug (0.12.2)
wheel (0.30.0)
wxPython (3.0.2.0)
wxPython-common (3.0.2.0)
xpcconnection (0.0.1)

ivankravets, 2017-10-17:
@japina please provide contents of `~/.bash_profile`. Your $PATH is empty, it looks that you broke system environment.

japina, 2017-10-18:
Ah, sorry .. it was late yesterday. I posted data from platformio terminal. System terminal is completely different.

1.
/Users/bostjan/.cargo/bin:/Library/Frameworks/Python.framework/Versions/3.5/bin:/usr/local/Cellar/e2fsprogs/1.42.13/sbin:/Users/bostjan/peskovnik/esp/xtensa-lx106-elf/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin:/usr/local/go/bin:/Library/TeX/texbin:/usr/local/gcc-arm-none-eabi-5_4-2016q2/bin:/Users/bostjan/bin

2. empty

3. Python 2.7.10
4. pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)
5.
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
arrow (0.10.0)
backports-abc (0.5)
bleep (0.4.4.dev3)
bottle (0.12.13)
certifi (2017.4.17)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
configobj (5.0.6)
Flask (0.12.2)
future (0.16.0)
glob2 (0.6)
idna (2.5)
ino (0.3.7)
itsdangerous (0.24)
Jinja2 (2.9.6)
lockfile (0.12.2)
MarkupSafe (1.0)
mercurial (4.3.2)
pip (9.0.1)
PyBluez (0.22)
pycairo (1.15.3)
pycam (0.6.1)
pygattosx (0.0.1)
pygobject (3.26.0)
pyobjc-core (4.0)
pyobjc-framework-Cocoa (4.0)
pyportmidi (0.0.7)
pyserial (3.2.1)
python-dateutil (2.6.0)
requests (2.18.1)
semantic-version (2.6.0)
setuptools (36.5.0)
singledispatch (3.4.0.3)
six (1.10.0)
tornado (4.5.2)
urllib3 (1.21.1)
virtualenv (15.1.0)
Werkzeug (0.12.2)
wheel (0.30.0)
wxPython (3.0.2.0)
wxPython-common (3.0.2.0)
xpcconnection (0.0.1)

paulo9999, 2017-10-26:
I'm following the discussion here, I'm having the same issue as @japina... 

my $PATH is empty
and I cannot find the folder ~/.bash_profile

MHesham, 2017-11-03:
I am facing the same issue on Ubuntu 16.04:
```
PlatformIO Plus (https://pioplus.com) v0.10.13
CorePackageManager: Installing contrib-piohome @ >=0.3.2,<2
Error: Detected unknown package 'contrib-piohome'
```
I can see this error in ATOM terminal:
`Uncaught (in promise) Could not start PIO Home server: Error: timeout`

ivankravets, 2017-11-03:
Fixed, please restart IDE.

closed 2017-11-03 by ivankravets

evoteknologi, 2017-11-03:
Why im getting this ? I have tried all method. Remove .platformio and install again, use buit in core, restart IDE, im using vsc dan atom.

[image]

ivankravets, 2017-11-03:
@evoteknologi could you provide full exception information?

evoteknologi, 2017-11-04:
@ivankravets 
[image]

[image]

My pio home always loading white background in atom, in vsc always loading also.

ivankravets, 2017-11-04:
@evoteknologi you have restrictions in a system which don't allow to run `pioplus.exe` binary which is responsible for PIO Home. Did you configure it manually?

evoteknologi, 2017-11-04:
@ivankravets not yet, how to do that sir?

evoteknologi, 2017-11-04:
@ivankravets now its worked just type pio home in command prompt as administrator. Im forget to run admin. Thank you very much sir

AlexFBP, 2017-12-26:
For me, worked installing pip as the last image by https://github.com/platformio/platformio-atom-ide/issues/918#issuecomment-341855801 

vscode from elementary 0.4 (ubuntu 16)

`pip install -U platformio`