platformio/platformio-atom-ide #929: InstallationManager

ashtez, 2017-09-11:
PIP: Error: spawn EPERM when trying to download core.

ivankravets, 2017-10-17:
Please remove `%HOME_DIR%/.platformio` and restart IDE

george-vieira, 2017-12-09:
I've done that and after starting the IDE, it goes to download the core again and I get the same error. .platformio gets recreated

ivankravets, 2017-12-11:
@george-vieira what is your OS?

george-vieira, 2017-12-12:
Windows 10 x64 Version	10.0.16299 Build 16299

ivankravets, 2017-12-12:
@george-vieira 
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

george-vieira, 2017-12-14:
Is the installation meant to install python automatically? It appears it didn't or didn't add the path. I've already rebooted so doesn't appear installed.

Microsoft Windows [Version 10.0.16299.19]
(c) 2017 Microsoft Corporation. All rights reserved.

C:\Users\george.vieira\Desktop>echo %PATH%
C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;D:\Program Files\OpenVPN\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files\LLVM\bin;C:\Users\george.vieira\AppData\Local\Microsoft\WindowsApps;C:\Users\george.vieira\AppData\Local\atom\bin

C:\Users\george.vieira\Desktop>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\george.vieira\Desktop>python --version
'python' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\george.vieira\Desktop>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\george.vieira\Desktop>pip list
'pip' is not recognized as an internal or external command,
operable program or batch file.

ivankravets, 2017-12-14:
1. Please open Windows Control Center and remove all Python applications from a system.
2. Remove `%HOME_DIR%/.platformio` folder
3. Start IDE

Does it work now?

george-vieira, 2017-12-14:
I removed python 2.7 and 3.6 as well as Python Launcher (no idea where that came from), killed the .platformio and restarted the app, same issue.

Just figured it out, Avast antivirus on windows was blocking it and I had the AV in Silent Mode (bloody annoying ads). Adding the path to exclusion list fixed it.

Might want to put a note on docs about possible PUP issues with antivirus software if enabled to either turn it off or exclude the .platformio\penv\*  paths

ivankravets, 2017-12-15:
Thank you for the report! We inform users about this issue.

closed 2017-12-15 by ivankravets