platformio/platformio-atom-ide #936: InstallationManager

abogdanovich, 2017-09-12:
Virtualenv Create: Traceback (most recent call last):
  File "/home/abogdnaovich/.platformio/.cache/tmp-3902jfNZX42nlGSw/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/home/abogdnaovich/.platformio/.cache/tmp-3902jfNZX42nlGSw/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/home/abogdnaovich/.platformio/.cache/tmp-3902jfNZX42nlGSw/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/abogdnaovich/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath
    if not isabs(path):
  File "/home/abogdnaovich/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-11-06:
Please reopen if you still need help

closed 2017-11-06 by ivankravets

abogdanovich, 2018-03-01:
Sorry guys - i'm out of my working place - so will try to reproduce asap. Thanks for understanding.