platformio/platformio-atom-ide #940: InstallationManager

DanStro, 2017-09-13:
Virtualenv Download: Traceback (most recent call last):
  File "C:\Users\dans\AppData\Local\Temp\tmp-9876E9sBi7hIDoYS\virtualenv-15.1.0\virtualenv.py", line 2328, in <module>
    main()
  File "C:\Users\dans\AppData\Local\Temp\tmp-9876E9sBi7hIDoYS\virtualenv-15.1.0\virtualenv.py", line 713, in main
    symlink=options.symlink)
  File "C:\Users\dans\AppData\Local\Temp\tmp-9876E9sBi7hIDoYS\virtualenv-15.1.0\virtualenv.py", line 945, in create_environment
    download=download,
  File "C:\Users\dans\AppData\Local\Temp\tmp-9876E9sBi7hIDoYS\virtualenv-15.1.0\virtualenv.py", line 901, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\dans\AppData\Local\Temp\tmp-9876E9sBi7hIDoYS\virtualenv-15.1.0\virtualenv.py", line 797, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\dans\.platf...v\Scripts\python.exe - setuptools pip wheel failed with error code 1

ivankravets, 2017-10-17:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

DanStro, 2017-10-17:
Hi Ivan,

Since I posted that message I've uninstalled platformio so I can't give you the information you're asking for.

Thanks though.

ivankravets, 2017-10-17:
Thanks! Don't hesitate to report next time. Sorry for the delay.

closed 2017-10-17 by ivankravets