platformio/platformio-atom-ide #999: InstallationManager- Please, help with this error..

christiamdc, 2017-09-29:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\christiam.munoz-del-\.platformio\.cache\tmp-9328IhmrXUJvgGAl\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\christiam.munoz-del-\.platformio\.cache\tmp-9328IhmrXUJvgGAl\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\christiam.munoz-del-\.platformio\.cache\tmp-9328IhmrXUJvgGAl\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\christiam.munoz-del-\.platformio\.cache\tmp-9328IhmrXUJvgGAl\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\christiam.munoz-del-\.platformio\.cache\tmp-9328IhmrXUJvgGAl\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\christiam.m...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

christiamdc, 2017-09-29:
C:\Users\christiam.munoz-del->python --version
Python 3.6.2

C:\Users\christiam.munoz-del->pip --version
pip 9.0.1 from c:\users\christiam.munoz-del-\appdata\local\programs\python\pytho
n36\lib\site-packages (python 3.6)

C:\Users\christiam.munoz-del->echo %path%
C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\Sy
stem32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\J
ava\jre7\bin\;C:\Program Files (x86)\nodejs\;C:\Program Files\Intel\WiFi\bin\;C:
\Program Files\Common Files\Intel\WirelessCommon\;C:\Windows\System32\WindowsPow
erShell\v1.0\;C:\Program Files (x86)\Skype\Phone\;C:\Users\christiam.munoz-del-\
AppData\Local\Programs\Python\Python36;C:\Users\christiam.munoz-del-\AppData\Loc
al\Programs\Python\Python36\Scripts;C:\Users\christiam.munoz-del-\AppData\Roamin
g\npm;C:\Users\christiam.munoz-del-\AppData\Local\atom\bin

ivankravets, 2017-10-17:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/703

closed 2017-10-17 by ivankravets