platformio/platformio-atom-ide #1005: Do not copy empty lines as line breaks.

kon-anast, 2017-10-01:
If you copy something and afterwords accidentally press copy again on an empty line, it will copy a line break and it will lose the previous copied lines or text. I do not think that it is necessary to copy just a line break.
I think it will be better if it keeps the previous copied text or lines if you press copy on an empty line. I could not find any useful reason to copy just line breaks. Correct me if I am wrong.

ivankravets, 2017-10-17:
Do you mean terminal?

konanast, 2017-10-19:
Thanks for the reply.
I do not mean in the terminal. I mean in the editor while you writing your code, if you are at an empty line and copy it, it will copy the empty line with the line break. This happens due to the fact that if you do not select a piece of text, and copy (for example with Ctrl+C), it will copy the whole line where the pointer is. That's great, but I think it is unnecessary to keep a copy of a  line break in the clipboard if the line is empty. I hope It's more clear now.
Maybe it is how atom works and not platformio.

ivankravets, 2017-10-19:
You can discuss it here https://github.com/atom/atom

closed 2017-10-19 by ivankravets