platformio/platformio-atom-ide #1012: InstallationManager

guigashm, 2017-10-06:
Virtualenv Create: Traceback (most recent call last):
  File "/Users/guigas/.platformio/.cache/tmp-38735IVBBpP6BT2j0/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/Users/guigas/.platformio/.cache/tmp-38735IVBBpP6BT2j0/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/Users/guigas/.platformio/.cache/tmp-38735IVBBpP6BT2j0/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/guigas/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 360, in abspath
    if not isabs(path):
  File "/Users/guigas/.platformio/penv/bin/../lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-10-17:
Could you try?
```
rm -rf /Users/guigas/.platformio/penv
pip install --user virtualenv
/Library/Python/2.7/bin/virtualenv /Users/guigas/.platformio/penv
source /Users/guigas/.platformio/penv/bin/activate
pip install platformio
```

Does it work?

jrial, 2017-10-27:
@ivankravets Not that exact sequence of commands, since for some reason the user install of virtualenv also errored out. But creating a scratch venv, and installing virtualenv within that venv, and using that to create the platformio venv did work (a bit of a convoluted approach).

For the record: my Linux mint has virtualenv at version 15.0.1, within the new venv it's version 15.1.0.

jrial, 2017-10-27:
Additional: This was on the console, but the error was thrown from the Atom plugin. While the console installation went fine, the Atom plugin never registered a successful install, so it retried and hit the same erorr upon restart. Turns out the Python 3 virtualenv package was the culprit.

This package depends on virtualenv, which in turn depends on python3-virtualenv. Turns out during the installation, this virtualenv was used to create the Platformio venv, which was Python 2.7-flavoured. And that somehow got things mixed up, resulting in an error message similar to the one this ticket describes.

Uninstalling these 3 packages, I was left with only python-virtualenv and virtualenv-clone on my system. Restarting Atom now resulted in a successful installation.

ivankravets, 2017-11-04:
@jrial Thanks a lot for your explanation!

closed 2017-11-04 by ivankravets