platformio/platformio-atom-ide #1028: InstallationManager: I can't get rid of this error.

dreamscaperia, 2017-10-14:
Virtualenv Create: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp-17772wqfDHniMsMqD\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\.platformio\.cache\tmp-17772wqfDHniMsMqD\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\.platformio\.cache\tmp-17772wqfDHniMsMqD\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\.platformio\.cache\tmp-17772wqfDHniMsMqD\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\.platformio\.cache\tmp-17772wqfDHniMsMqD\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\.platformio\penv\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

dreamscaperia, 2017-10-14:
Now I seem to figure out what exactly this is from. My Windows 10 uses "GBK" as default decoding. 
So when 'virtualenv.py' tries to install wheel in line 877, it will fail with no doubt.
(You can't even install wheel to your own Python Environment, when your system uses a default decoding other than "UTF-8".)

Here is the proof.
Try the following:
1. close Atom/VSCode after the error happens.
2. mannually add the file 'sitecustomize.py' with the codeline "**sys.setdefaultencoding('gbk')**"  to "C:\.platformio\penv\Lib\site-packages".
3. Unzip the existing file "C:\.platformio\.cache\virtualenv.tar.gz" in place.
4. execute this cmdline "python virtualenv.py c:\.platformio\penv" in dir "c:\.platformio\.cache\virtualenv-14.0.6".
Then, you will see the output "Installing setuptools, pip, wheel...done." with no errors.

Though I figure it out, But I can't solve it! because THE NEXT TIME I open the IDE, the installation will restart the procedure and then DELETE THE ENTIRE DIR "c:\.platformio\penv"! Then my effort VANISHES!
Unless I can make WIndows to decode "UTF-8" by default or prevent the installation from deleting the directory, This issue can't be fixed. (Obviously I can't make both of them happen...)

Let me know if I'm wrong.

dreamscaperia, 2017-10-14:
Wow! I then tried to insert the file 'customize.py' to the dir as fast as I could after the procedure deleted the dir and then recreated it, using this cmdline "copy sitecustomize.py C:\.platformio\penv\Lib\site-packages".
Guess what? I made it!
Now the platform-ide finally runs inside my VSCode!!!

dreamscaperia, 2017-10-14:
Anyway, such a buggy product. I have every right to express my anger.

dreamscaperia, 2017-10-14:
Anyway, platformIO is such a buggy product. I have every right to express my anger.

ivankravets, 2017-10-17:
Please provide an output of 
```
python -c "import sys; print sys.getdefaultencoding();"
python -c "import sys; print sys.stdin.encoding"
```

Thanks!

ivankravets, 2017-10-17:
What is your locale?

[image]

dreamscaperia, 2017-10-18:
Here is the output. My system does use "GBK": 

C:\Users\xxx>python -c "import sys; print sys.getdefaultencoding();"
gbk
C:\Users\xxx>python -c "import sys; print sys.stdin.encoding"
cp936

And I wouldn't change my Region to other Locale... I would miss the lovely Cortana...

ivankravets, 2017-10-18:
I’m not Windows expert. I run it in VM. 

Do you know how I can change locale to gbk?

Also, could you try this in CMD
```
pip install virtualenv 
virtualenv %HOME_DIR/.platformio/penv
%HOME_DIR/.platformio/penv/Scripts/activate
pip install platformio
```

Does it work?

dreamscaperia, 2017-10-18:
Nah. It didn't work out.

The second cmdline "virtualenv %HOME_DIR/.platformio/penv" gave the same errors (error code 2) and interrupt everything comes next.

Then I figured it out that the dir "penv" is created by virtualenv... So the dir preparation is part of virtualenv, not platformIO... While virtualenv is a package held by others, the problem becomes tough.

Because this error happens when virtualenv tries to install wheel in the virtual environment. And this operation is managed by the extraction of virtualenv. And the extraction of virtualenv cannot be altered ahead.

-------------------

And if you wanna change locale to "gbk" for testing, perhaps go find and select the option "Chinese(China)" or anything like that, in the drop-down list exactly in your screenshot, plus a reboot.

I'm not sure but I think this is it LOL.

Friendly provide: https://www.isunshare.com/windows-10/change-system-locale-in-windows-10.html

ivankravets, 2017-10-18:
Did you replace %HOME_DIR% with a real full path?

ivankravets, 2017-10-18:
I changed locale but there is no effect. Could you repeat https://github.com/platformio/platformio-atom-ide/issues/1028#issuecomment-337605875 with full path?

dreamscaperia, 2017-10-19:
Maybe the Location Tab of Region should also be changed, too.

---------------------------------------------------

Here is the full output of mine that you may need:

> C:\>pip install virtualenv
> Requirement already satisfied: virtualenv in d:\python27\lib\site-packages
> 
> C:\>virtualenv c:\penv
> New python executable in c:\penv\Scripts\python.exe
> Installing setuptools, pip, wheel...
>   Complete output from command c:\penv\Scripts\python.exe - setuptools pip wheel:
>   Collecting setuptools
> Exception:
> Traceback (most recent call last):
>   File "d:\python27\lib\site-packages\virtualenv_support\pip-9.0.1-py2.py3-none-any.whl\pip\basecommand.py", line 215, in main
>     status = self.run(options, args)
>   File "d:\python27\lib\site-packages\virtualenv_support\pip-9.0.1-py2.py3-none-any.whl\pip\commands\install.py", line 324, in run
>     requirement_set.prepare_files(finder)
>   File "d:\python27\lib\site-packages\virtualenv_support\pip-9.0.1-py2.py3-none-any.whl\pip\req\req_set.py", line 380, in prepare_files
>     ignore_dependencies=self.ignore_dependencies))
>   File "d:\python27\lib\site-packages\virtualenv_support\pip-9.0.1-py2.py3-none-any.whl\pip\req\req_set.py", line 620, in _prepare_file
>     session=self.session, hashes=hashes)
>   File "d:\python27\lib\site-packages\virtualenv_support\pip-9.0.1-py2.py3-none-any.whl\pip\download.py", line 821, in unpack_url
>     hashes=hashes
>   File "d:\python27\lib\site-packages\virtualenv_support\pip-9.0.1-py2.py3-none-any.whl\pip\download.py", line 659, in unpack_http_url
>     hashes)
>   File "d:\python27\lib\site-packages\virtualenv_support\pip-9.0.1-py2.py3-none-any.whl\pip\download.py", line 880, in _download_http_url
>     file_path = os.path.join(temp_dir, filename)
>   File "c:\penv\lib\ntpath.py", line 85, in join
>     result_path = result_path + p_path
> UnicodeDecodeError: 'ascii' codec can't decode byte 0xb0 in position 7: ordinal not in range(128)
> ----------------------------------------
> ...Installing setuptools, pip, wheel...done.
> Traceback (most recent call last):
>   File "d:\python27\lib\runpy.py", line 174, in _run_module_as_main
>     "__main__", fname, loader, pkg_name)
>   File "d:\python27\lib\runpy.py", line 72, in _run_code
>     exec code in run_globals
>   File "D:\Python27\Scripts\virtualenv.exe\__main__.py", line 9, in <module>
>   File "d:\python27\lib\site-packages\virtualenv.py", line 713, in main
>     symlink=options.symlink)
>   File "d:\python27\lib\site-packages\virtualenv.py", line 945, in create_environment
>     download=download,
>   File "d:\python27\lib\site-packages\virtualenv.py", line 901, in install_wheel
>     call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
>   File "d:\python27\lib\site-packages\virtualenv.py", line 797, in call_subprocess
>     % (cmd_desc, proc.returncode))
> OSError: Command c:\penv\Scripts\python.exe - setuptools pip wheel failed with error code 2
> 
> C:\>c:\penv\Scripts\activate
> 'c:\penv\Scripts\activate' 不是内部或外部命令，也不是可运行的程序
> 或批处理文件。
> 
> C:\>

ivankravets, 2017-11-06:
Could you re-test with the latest 2.0.0-rc.4 version? Does it work now?

closed 2017-11-07 by ivankravets