platformio/platformio-atom-ide #1043: 'PlatformIO Home' doesn't open

Spongman, 2017-10-17:
i just updated everything.

all i get when starting atom is this:

[image]

ivankravets, 2017-10-17:
Could you open in PIO IDE Terminal
```
pio home
```

Does it work?

beegee-tokyo, 2017-11-03:
Same here.
When I use 'pio home' in PIO IDE Terminal I get
```
PS C:\Users\beegee> pio home
PlatformIO Plus (https://pioplus.com) v0.10.13
CorePackageManager: Installing contrib-piohome @ >=0.3.2,<2
Error: Detected unknown package 'contrib-piohome'
```

beegee-tokyo, 2017-11-03:
Solved that problem with downloading the contrib-piohome package manually and unpack it in packages folder. There is no version 0.3.2 available on bintray, only 0.3.1, I guess that might be the problem because in core.py it requests version 0.3.2.

ivankravets, 2017-11-03:
Fixed, just run `pio update`. Should be PIO Home 0.3.2

Spongman, 2017-11-04:
i did an update, including `pio update`, and after restart i get:
````
PS C:\Users\User> pio home
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 21, in init cmd_debug
  File "C:\Users\User\.platformio\packages\pysite-pioplus\twisted\internet\stdio.py", line 30, in <module>
    from twisted.internet import _win32stdio
  File "C:\Users\User\.platformio\packages\pysite-pioplus\twisted\internet\_win32stdio.py", line 9, in <module>
    import win32api
ImportError: DLL load failed: %1 is not a valid Win32 application.
````

Spongman, 2017-11-04:
ok, i uninstalled, reinstall python & platform, now i get this:
````
PS C:\Users\PiersH> pio update
Updating tool-scons                      @ 3.20501.2      [Up-to-date]
Updating tool-unity                      @ 1.20302.1      [Up-to-date]
Updating pysite-pioplus                  @ 0.4.2          [Up-to-date]
Updating tool-pioplus                    @ 0.10.13        [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------

Updating atmelavr                        @ 1.7.1          [Up-to-date]
Updating toolchain-atmelavr              @ 1.40902.0      [Up-to-date]
Updating framework-arduinoavr            @ 1.10619.0      [Up-to-date]
Updating tool-avrdude                    @ 1.60300.2      [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ 0.10.0         [Up-to-date]
Updating tool-esptoolpy                  @ 1.20000.0      [Up-to-date]
Updating toolchain-xtensa32              @ 1.50200.0      [Up-to-date]
Updating framework-arduinoespressif32    @ 1.2.0          [Up-to-date]
Updating framework-espidf                @ 2.210.0        [Up-to-date]
Updating tool-espotapy                   @ 1.0.0          [Up-to-date]

Library Manager
===============
PS C:\Users\PiersH> pio home
Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\users\piersh\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\piersh\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\piersh\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\piersh\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
````

closed 2017-11-07 by ivankravets