platformio/platformio-atom-ide #1056: Installer assumes that conda install python -> a working pip

pelson, 2017-10-21:
cecd5554f1031f8390ea0daa5cb5c368b4065412 introduced a commit to allow conda support. I have conda installed, and it is getting used. Unfortunately, I also have conda configured to not automatically install pip when I install python. If I want pip installed, I need to state it explicitly. I'd encourage the same in the platformio ide codebase:

```
conda install python=2.7 pip ...
```

Should be all that is necessary.

pelson, 2017-10-21:
Incidentally, without this, I was getting Installation manager errors:

```
PIP: Error: spawn /Users/pelson/.platformio/penv/bin/pip ENOENT
```

ivankravets, 2017-11-06:
Please remove `/Users/pelson/.platformio/penv`, then
1. Create virtualenv in `/Users/pelson/.platformio/penv` with PIP
2. Activate virtual environment and install `pip install platformio`.

Restart IDE.

pelson, 2017-11-08:
Thanks for responding, but I think you missed the important detail here: I'm raising a bug report. platformio's use of conda is making an incorrect assumption. It is easy enough to fix (and I'm happy to do so) if only I could find the right bit of the code to modify. Essentially, the addition of pip as a dependency when making the ``conda create`` call is the only thing that needs to change.

pelson, 2017-11-08:
The commit in question: https://github.com/platformio/platformio-atom-ide/commit/cecd5554f1031f8390ea0daa5cb5c368b4065412

Essentially, you have a line that says:

```
 utils.runCommand(
         'conda',
         ['create', '--yes', '--quiet', 'python=2', '--prefix', config.ENV_DIR],
```

This needs to be updated to:

```
 utils.runCommand(
        'conda',
        ['create', '--yes', '--quiet', 'python=2', 'pip', '--prefix', config.ENV_DIR],
```

Unfortunately the code has moved a bit since that commit, and I can't actually find where conda is being used anymore (maybe it isn't?)

ivankravets, 2017-11-08:
Will it work only for you or for other Conda's users?

pelson, 2017-11-08:
> Will it work only for you or for other Conda's users?

Everybody. The key is that a conda environment is being created, and subsequently the assumption is that the environment contains pip. Therefore we should state that pip should be installed in the environment explicitly.

(FWIW I have a lot of conda experience, and would be happy to help if ever conda questions come up...)

pelson, 2017-11-08:
P.S. Thanks for re-opening. I would normally just open a PR, but didn't have much of a chance to find my way around the project and get a development installation.

ivankravets, 2017-11-08:
See https://github.com/platformio/platformio-node-helpers/blob/master/src/installer/stages/platformio-core.js#L149

closed 2017-11-08 by ivankravets