platformio/platformio-atom-ide #1088: C/C++ project index rebuild failed

malefs, 2017-10-28:
Error: Traceback (most recent call last):
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/commands/init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 39, in __init__
    self._gather_tplvars()
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 152, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/util.py", line 134, in __call__
    value = self.func(*args)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 74, in get_project_build_data
    target=["idedata"])
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/commands/run.py", line 109, in cli
    result = (envname, ep.process())
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/commands/run.py", line 193, in process
    result = self._run()
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/commands/run.py", line 292, in _run
    skip_default_package=True)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/commands/platform.py", line 304, in platform_install
    skip_default_package=skip_default_package):
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/platform.py", line 68, in install
    self, name, requirements, silent=silent)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/package.py", line 628, in install
    name, url, requirements, track=True)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/package.py", line 368, in _install_from_url
    vcs = VCSClientFactory.newClient(tmp_dir, url)
  File "/home/lena/.platformio/penv/local/lib/python2.7/site-packages/platformio/vcsclient.py", line 42, in newClient
    obj = getattr(modules[__name__], "%sClient" % type_.title())(
AttributeError: 'module' object has no attribute 'Espressif8266HttpClient'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-11-04:
Could you provide your `platformio.ini`?

ivankravets, 2017-11-04:
It seems that INI file is broken. See http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps

closed 2017-11-04 by ivankravets