platformio/platformio-atom-ide #1090: InstallationManager

desperado88, 2017-10-28:
Virtualenv Create: Traceback (most recent call last):
  File "/Volumes/Macintosh HD/mathieu/.platformio/.cache-ide/tmp-3442VWVMYX2VsjDN/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>

    main()
  File "/Volumes/Macintosh HD/mathieu/.platformio/.cache-ide/tmp-3442VWVMYX2VsjDN/virtualenv-14.0.6/virtualenv.py", line 705, in main

    symlink=options.symlink)
  File "/Volumes/Macintosh HD/mathieu/.platformio/.cache-ide/tmp-3442VWVMYX2VsjDN/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment

    download=download,
  File "/Volumes/Macintosh HD/mathieu/.platformio/.cache-ide/tmp-3442VWVMYX2VsjDN/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel

    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/Volumes/Macintosh HD/mathieu/.platformio/.cache-ide/tmp-3442VWVMYX2VsjDN/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess

    % (cmd_desc, proc.returncode))

OSError: Command "/Volumes/Macintosh H...rmio/penv/bin/python" -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-11-06:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

desperado88, 2017-11-06:
Hello,

The Terminal returns this :

mathieu@ ~ $ echo $PATH
/opt/local/bin:/opt/local/sbin:/Library/Frameworks/Python.framework/Versions/3.4/bin:/Library/Frameworks/Python.framework/Versions/3.5/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin
mathieu@ ~ $ echo $PYTHONPATH

mathieu@ ~ $ python --version
Python 2.7.10
mathieu@ ~ $ pip --version
-bash: pip: command not found
mathieu@ ~ $ pip list
-bash: pip: command not found

ivankravets, 2017-11-06:
What is your Mac OS version?

ivankravets, 2017-11-06:
Are you sure?

closed 2017-12-01 by ivankravets