platformio/platformio-atom-ide #1123: C/C++ project index rebuild failed

Heppokoyuki, 2017-11-06:
Error: [Mon Nov  6 17:17:18 2017] Processing esp_wroom_02 (platform: espressif8266; board: esp_wroom_02; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PlatformioException: Could not load broken JSON: /home/yuki/.platformio/lib/ESP8266wifi-master/library.json:
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 156:
env.SConscript("$BUILD_SCRIPT")
File "/home/yuki/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/yuki/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/yuki/.platformio/platforms/espressif8266/builder/main.py", line 357:
target_elf = env.BuildProgram()
File "/home/yuki/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 64:
deplibs = env.BuildProjectLibraries()
File "/home/yuki/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 676:
lib_builders = env.GetLibBuilders()
File "/home/yuki/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 648:
env, join(libs_dir, item), verbose=verbose)
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 50:
env, path, verbose=verbose)
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 95:
self._manifest = manifest if manifest else self.load_manifest()
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 485:
manifest = util.load_json(join(self.path, "library.json"))
File "/home/yuki/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 192:
"Could not load broken JSON: %s" % file_path)
========================== [ERROR] Took 0.55 seconds ==========================

ivankravets, 2017-11-06:
Could you share a content of `/home/yuki/.platformio/lib/ESP8266wifi-master/library.json`?

ivankravets, 2017-11-07:
Please reopen if you still need help

closed 2017-11-07 by ivankravets

MortyFleem, 2023-05-02:
Same problem. Cannot find solution on this page.