platformio/platformio-atom-ide #1131: InstallationManager

bdurrer, 2017-11-07:
```
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Beat\.platformio\.cache\tmp-6880dx7EkpFDeUvP\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Beat\.platformio\.cache\tmp-6880dx7EkpFDeUvP\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Beat\.platformio\.cache\tmp-6880dx7EkpFDeUvP\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Beat\.platformio\.cache\tmp-6880dx7EkpFDeUvP\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\Beat\.platformio\.cache\tmp-6880dx7EkpFDeUvP\virtualenv-14.0.6\virtualenv.py", line 743, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Python27\Lib\subprocess.py", line 710, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 958, in _execute_child
    startupinfo)
WindowsError: [Error 5] Zugriff verweigert
```

Starting Atom as administrator does not help.
It occurs on every startup, while that little "Plattform IDE: Installing..." window is shown.

bdurrer, 2017-11-07:
Oh, and it's Windows 10.

ivankravets, 2017-11-07:
Python can't run subprocess. It seems that you have specific rules in OS. Try to run Atom with Administrative permissions or tune existing

bdurrer, 2017-11-07:
I did run it with as admin, which I stated. 
That was really not helpful at all, thanks for nothing.

ivankravets, 2017-11-07:
Do you have antivirus in a system?

closed 2017-11-07 by bdurrer

ivankravets, 2017-11-07:
Ah... Just try to remove `%HOME_DIR%/.platformio/penv` folder and start build again.

Also, take a look at PIO + VSCode for embedded programming.

http://platformio.org/get-started/ide?install=vscode