platformio/platformio-atom-ide #1145: C/C++ project index rebuild failed

Resinder, 2017-11-13:
Error: [Mon Nov 13 22:17:47 2017] Processing nanoatmega328 (platform: atmelavr; board: nanoatmega328; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PlatformioException: Could not load broken JSON: /home/master/.platformio/lib/HTU21D/library.json:
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/main.py", line 156:
env.SConscript("$BUILD_SCRIPT")
File "/home/master/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/master/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/master/.platformio/platforms/atmelavr/builder/main.py", line 218:
target_elf = env.BuildProgram()
File "/home/master/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 64:
deplibs = env.BuildProjectLibraries()
File "/home/master/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 676:
lib_builders = env.GetLibBuilders()
File "/home/master/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 648:
env, join(libs_dir, item), verbose=verbose)
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 50:
env, path, verbose=verbose)
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 95:
self._manifest = manifest if manifest else self.load_manifest()
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 485:
manifest = util.load_json(join(self.path, "library.json"))
File "/home/master/.platformio/penv/local/lib/python2.7/site-packages/platformio/util.py", line 192:
"Could not load broken JSON: %s" % file_path)
======================================================================= [ERROR] Took 0.37 seconds =======================================================================

Resinder, 2017-11-13:
Help me please. I need to learn using another ide to compile ini or ino files and edit them by using platformio ide editor. Thank you for all of your help and kind behaviour.

ivankravets, 2017-12-01:
Please fix this JSON file /home/master/.platformio/lib/HTU21D/library.json. Where did you find this library?

closed 2017-12-01 by ivankravets