platformio/platformio-atom-ide #1164: InstallationManager fails repeatedly

idesignstuff, 2017-11-22:
Error: Failed to download file https://pypi.python.org/packages/source/v/virtualenv/virtualenv-14.0.6.tar.gz: Error: unable to verify the first certificate
    at _callee$ (C:\Users\weeat\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2522:19)
    at tryCatch (C:\Users\weeat\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:780:40)
    at Generator.invoke [as _invoke] (C:\Users\weeat\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1014:22)
    at Generator.prototype.(anonymous function) [as throw] (C:\Users\weeat\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:832:21)
    at step (C:\Users\weeat\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2631:191)
    at C:\Users\weeat\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:2631:402
    at process._tickCallback (internal/process/next_tick.js:103:7)

ivankravets, 2017-12-01:
Could you restart IDE? Something is wrong with your certificates chain.

ivankravets, 2017-12-01:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

idesignstuff, 2017-12-02:
C:\>echo %PATH%
C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\LLVM\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Zerynth\ztc\windows64;C:\Program Files\Git\cmd;C:\Users\weeat\.platformio\python27\Scripts;C:\Users\weeat\AppData\Local\Microsoft\WindowsApps;C:\Users\weeat\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Microsoft VS Code\bin;C:\Users\weeat\AppData\Local\atom\bin;C:\Python27;C:\Python27\Lib\site-packages\;C:\Python27\Scripts\;

C:\>echo %PYTHONPATH%
%PYTHONPATH%

C:\>python --version
Python 2.7.14

C:\>pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

C:\>pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
pip (9.0.1)
setuptools (28.8.0)

idesignstuff, 2017-12-02:
I think the problem is an incompatibility between python/pip/npm and covenant eyes (web filter and monitoring service, using man-in-the-middle routing).

ivankravets, 2017-12-02:
There is a problem with Node.JS which can't download a file. But you can do that manually via `pip install virtualenv`. Then restart IDE

closed 2017-12-15 by ivankravets