platformio/platformio-atom-ide #1171: C/C++ project index rebuild failed

thelastgamer, 2017-11-23:
Error: Could not install 'toolchain-gccarmnoneeabi' with version requirements '>=1.60301.0' for your system 'windows_x86'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

ivankravets, 2017-12-27:
@JenyaZolotarev , @thelastgamer

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ShardanX, 2018-01-02:
Nope.
Even deinstalling the Antivir and deactivating local firewall completely does definitely NOT help!
Tested this on three PCs now.
I've even tried to re-install atom + platformio completely fresh.
No Chance. Now i get a "C/C++ project index rebuild failed " on top.

ivankravets, 2018-01-10:
@JenyaZolotarev @thelastgamer @ShardanX  @kristsm

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

anishkgt, 2018-02-03:
i get the same error. I was just testing the Atom IDE and adding the platformio-ide gave this error

> [Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.1

### Stack Trace

Uncaught Error: Could not start PIO Home server: Error: Could not install 'contrib-pysite' with version requirements '>=0.1.2,<2' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

```
At C:\Users\georget\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1616

Error: Could not start PIO Home server: Error: Could not install 'contrib-pysite' with version requirements '>=0.1.2,<2' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1616:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:538:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at ChildProcess.cp.emit (/packages/platformio-ide/node_modules/cross-spawn/lib/enoent.js:34:29)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.16 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.1 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
tool-bar 1.1.4 
```

[image]

anishkgt, 2018-02-03:
when trying to built a sketch I  ge
[image]
t this as well

kobidkunda, 2020-06-07:
I solved by installing this `https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/macos-setup.html#toolchain-setup`

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets