platformio/platformio-atom-ide #1175: InstallationManager

vedatAltun, 2017-11-24:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\user\.platformio\.cache\tmp-5784GbWM47bka5PA\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\user\.platformio\.cache\tmp-5784GbWM47bka5PA\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\user\.platformio\.cache\tmp-5784GbWM47bka5PA\virtualenv-14.0.6\virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "C:\Users\user\.platformio\.cache\tmp-5784GbWM47bka5PA\virtualenv-14.0.6\virtualenv.py", line 1197, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Python27\lib\shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\user\\.platformio\\penv\\Scripts\\python.exe'

ivankravets, 2017-12-15:
Please reopen if you still need help

closed 2017-12-15 by ivankravets

vedatAltun, 2017-12-15:
Thanks  I solved  my problem

ivankravets, 2017-12-15:
How did you solve it?