platformio/platformio-atom-ide #1179: InstallationManager

frdteknikelektro, 2017-11-24:
Conda Virtualenv: An unexpected error has occurred.
Please consider posting the following information to the
conda GitHub issue tracker at:

    https://github.com/conda/conda/issues

Current conda install:

               platform : linux-64
          conda version : 4.3.21
       conda is private : False
      conda-env version : 4.3.21
    conda-build version : not installed
         python version : 2.7.13.final.0
       requests version : 2.14.2
       root environment : /home/farid/anaconda2  (writable)
    default environment : /home/farid/anaconda2
       envs directories : /home/farid/anaconda2/envs
                          /home/farid/.conda/envs
          package cache : /home/farid/anaconda2/pkgs
                          /home/farid/.conda/pkgs
           channel URLs : https://repo.continuum.io/pkgs/free/linux-64
                          https://repo.continuum.io/pkgs/free/noarch
                          https://repo.continuum.io/pkgs/r/linux-64
                          https://repo.continuum.io/pkgs/r/noarch
                          https://repo.continuum.io/pkgs/pro/linux-64
                          https://repo.continuum.io/pkgs/pro/noarch
            config file : None
             netrc file : None
           offline mode : False
             user-agent : conda/4.3.21 requests/2.14.2 CPython/2.7.13 Linux/4.4.0-101-generic debian/stretch/sid glibc/2.23    
                UID:GID : 1000:1000

`$ /home/farid/anaconda2/bin/conda create --yes --quiet python=2 --prefix /home/farid/.platformio/penv`

    Traceback (most recent call last):
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/exceptions.py", line 632, in conda_exception_handler
        return_value = func(*args, **kwargs)
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/cli/main.py", line 137, in _main
        exit_code = args.func(args, p)
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/cli/main_create.py", line 68, in execute
        install(args, parser, 'create')
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/cli/install.py", line 357, in install
        execute_actions(actions, index, verbose=not context.quiet)
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/plan.py", line 829, in execute_actions
        with History(actions[PREFIX]):
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/history.py", line 72, in __enter__
        self.update('enter')
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/history.py", line 91, in update
        self.init_log_file()
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/history.py", line 81, in init_log_file
        self.write_dists(linked(self.prefix))
      File "/home/farid/anaconda2/lib/python2.7/site-packages/conda/history.py", line 255, in write_dists
        os.makedirs(self.meta_dir)
      File "/home/farid/anaconda2/lib/python2.7/os.py", line 157, in makedirs
        mkdir(name, mode)
    OSError: [Errno 17] File exists: '/home/farid/.platformio/penv/conda-meta'

ivankravets, 2017-12-01:
Could you remove this folder `/home/farid/.platformio/penv` and restart IDE? Does it work now?

frdteknikelektro, 2017-12-02:
Sorry, I don't know how, but it work even I didn't do anything. Just restart and reinstalling. Sorry forgot to closing this.

closed 2017-12-02 by frdteknikelektro