platformio/platformio-atom-ide #1184: C/C++ project index rebuild failed

DjamesSuhanko, 2017-11-25:
Error: spawn platformio ENOENT
When downloading a new core, clicking on item blinking.

ivankravets, 2017-12-01:
What is your OS?

http://docs.platformio.org/en/latest/faq.html#platformio-is-not-recognized-as-an-internal-or-external-command

DjamesSuhanko, 2017-12-01:
Hello! I'm using Linux. For now, I'm using PlatformIO with CodeBlocks. The next, I will try with QtCreator too. The problem was not so bad, I could learn more about PlatformIO. But I think peoples will not try another solution because any other one needs command line.
Thanks to answer me!

ivankravets, 2017-12-01:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

DjamesSuhanko, 2017-12-01:
$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/djames/esp/xtensa-esp32-elf/bin:/home/djames/bin:/home/djames/opt/qcad-3.4.7-pro-linux-x86_64:/usr/local/ant/bin

$ echo $PYTHONPATH (nothing here)

$ which python
/usr/bin/python
$ pip --version
pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

$ pip list
acora (2.0)
alabaster (0.7.8)
arduino-helpers (0.3.post18)
arduino-scons (0.1.post11)
argcomplete (1.8.1)
argh (0.26.2)
arrow (0.10.0)
artifacts (20170808)
attrs (17.2.0)
Babel (2.4.0)
backports.shutil-get-terminal-size (1.0.0)
beautifulsoup4 (4.6.0)
binplist (0.1.5)
BitTorrent (3.4.2)
blessings (1.6)
bottle (0.12.13)
bpython (0.17)
capstone (3.0.4)
chardet (3.0.4)
Cheetah (2.4.4)
click (5.1)
colorama (0.3.7)
configparser (3.5.0)
constantly (15.1.0)
construct (2.5.2)
cryptography (1.7.1)
curtsies (0.2.11)
cycler (0.10.0)
decorator (4.1.2)
dfdatetime (20170719)
dfvfs (20170723)
dfwinreg (20170706)
dissy (9)
distorm3 (3.3.4)
docutils (0.14)
dpkt (1.8)
dumbnet (1.12)
efilter (1!1.5)
enum34 (1.1.6)
functools32 (3.2.3.post2)
future (0.15.2)
greenlet (0.4.12)
hachoir-core (1.3.3)
hachoir-metadata (1.3.3)
hachoir-parser (1.3.4)
html5lib (0.999999999)
idna (2.5)
imagesize (0.7.1)
impacket (0.9.15)
incremental (16.10.1)
intervaltree (2.1.0)
ipaddr (2.1.11)
ipaddress (1.0.17)
IPy (0.83)
ipython (5.5.0)
ipython-genutils (0.2.0)
isodate (0.5.4)
jdcal (1.0)
Jinja2 (2.9.6)
keyring (10.4.0)
keyrings.alt (2.2)
lockfile (0.12.2)
lxml (4.1.0)
MarkupSafe (1.0)
matplotlib (2.0.0)
Mirage (0.9.5.2)
mod-python (3.3.1)
mod-pywebsocket (0.8)
mysqlclient (1.3.10)
netifaces (0.10.4)
networkx (1.11)
numpy (1.13.1)
openpyxl (2.3.0)
PAM (0.4.2)
pandas (0.20.2)
paramiko (2.0.0)
path-helpers (0.4.post2)
pathlib2 (2.3.0)
pathtools (0.1.2)
pcapy (0.10.8)
pefile (2017.9.3)
pexpect (4.2.1)
pickleshare (0.7.4)
Pillow (4.0.0)
pip (9.0.1)
plaso (1.5.1)
platformio (3.4.1)
prompt-toolkit (1.0.14)
protobuf (3.0.0)
psutil (5.0.1)
psycopg2 (2.7.3)
py (1.4.34)
pyaff4 (0.24.post1)
pyasn1 (0.1.9)
pyasn1-modules (0.0.7)
PyChromecast (0.8.1)
pycrypto (2.6.1)
pycurl (7.43.0)
pydns (2.3.6)
pyelftools (0.24)
Pygments (2.2.0)
pygobject (3.24.1)
pygraphviz (1.4rc1)
PyOpenGL (3.1.0)
pyOpenSSL (16.2.0)
pyparsing (2.1.10)
pyrit (0.4.0)
PyScons (1.0.91)
pyserial (3.4)
pysnmp (4.3.2)
pysnmp-apps (0.3.2)
pysnmp-mibs (0.1.3)
pytest (3.2.1)
python-dateutil (2.6.1)
pytsk3 (20170802)
pytz (2017.2)
pyxdg (0.25)
PyYAML (3.12)
pyzmq (16.0.2)
rdflib (4.2.1)
rekall-core (1.6.0)
requests (2.12.4)
roman (2.0.0)
scandir (1.6)
scapy (2.3.3)
scipy (0.19.1)
scons (2.5.1)
sconsduino (0.2)
scour (0.32)
SecretStorage (2.3.1)
semantic-version (2.6.0)
serial-device (0.4.post4)
service-identity (16.0.0)
setuptools (36.6.0)
simplegeneric (0.8.1)
six (1.11.0)
sortedcontainers (1.5.7)
SPARQLWrapper (1.7.6)
Sphinx (1.4.9)
SQLAlchemy (1.1.11)
subprocess32 (3.2.7)
traitlets (4.3.2)
Twisted (16.6.0)
urllib3 (1.19.1)
urwid (1.3.1)
vboxapi (1.0)
vinetto (0.7b0)
virtualenv (15.1.0)
volatility (2.6)
watchdog (0.8.3)
wcwidth (0.1.7)
webencodings (0.5)
websocket-client (0.37.0)
wheel (0.29.0)
wxPython (3.0.2.0)
wxPython-common (3.0.2.0)
XlsxWriter (0.9.6)
yara-python (3.6.1)
zeroconf (0.19.1)
zope.interface (4.3.2)

ivankravets, 2017-12-01:
1. `which platformio`
2. `python --version`
3. `pio --version`

ivankravets, 2017-12-01:
http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

Please `pip uninstall platformio` in SYSTEM TERMINAL (NOT IDE). After that, all should work.

closed 2017-12-01 by ivankravets

DjamesSuhanko, 2017-12-02:
Ivan, I has installed platformio from pip because Atom was not working, so I tried another IDE. I did as you said, just to execute the procedure and test it again, but this procedure didn't solve the problem.
Don't worry, man. I'm using another IDE, but if you wish debugging, I'm at your disposal to make any test more.

ivankravets, 2017-12-02:
You didn't answer me 2nd time. What is your Python version? Also, what is your distro?

Take a look at this solution https://github.com/platformio/platformio-atom-ide/issues/333#issuecomment-345177760

DjamesSuhanko, 2017-12-02:
Sorry, was my fault, I forgot it. Please, be patience.
My distro is Debian Stretch.
My Python version:
Python 2.7.14
I have platformio here:
./.platformio/penv/bin/platformio
When installed from pip, of course, the path is another one:
/usr/local/bin/platformio - but as I said, removed from make the tests.
pio --version does not work. So I did:
./.platformio/penv/bin/platformio --version
PlatformIO, version 3.5.0rc4

ivankravets, 2017-12-02:
Do you use PlatformIO IDE Terminal?

DjamesSuhanko, 2017-12-02:
I'm using only after found the problem with Atom. Before, I was using PlatformIO-IDE, but platformio from terminal was needed to create the project using CodeBlocks as IDE.

ivankravets, 2017-12-02:
http://docs.platformio.org/en/latest/ide/atom.html#install-shell-commands

Also, take a look at our extension for VSCode. It's much better than Atom.

ivankravets, 2017-12-02:
http://docs.platformio.org/en/latest/ide/vscode.html

DjamesSuhanko, 2017-12-02:
Wow, good tip!
Thanks a lot, Ivan!