platformio/platformio-atom-ide #1189: Won't work: "C/C++ project index rebuild failed"

binderth, 2017-11-25:
I tried different Things explained in some issues here, including:
* deleting the `penv` Directory
* deactivating AntiVirus
* deleting the `.platformio`directory and installing again

in the end:
* deinstalling Atom
* deleting `%HOMEDIR%/.atom`
* deleting `%HOMEDIR%/.platformio`
* reinstalling Atom
* reinstalling platformio package

still the same issue. plattformioIO Core keeps update *everytime* i restart Atom. If I ignore it, the error keeps popping up. How and what can I provide for more information?

binderth, 2017-11-25:
After changing from Atom to VSCode, the same reinstalling/restarting procedure is present:
[image]
You can Keep klicking "Reload now", it will reload, then tell you, PlattformIO Core will be installed and then you can Klick "Reload now", and so on and so forth....

ivankravets, 2017-11-26:
Please upgrade to 0.8.0. Should be fixed.

closed 2017-11-26 by ivankravets