platformio/platformio-atom-ide #1197: InstallationManager

akhavea, 2017-11-25:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\akhav\.platformio\.cache\tmp-2199648I3sw0Wal7z\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\akhav\.platformio\.cache\tmp-2199648I3sw0Wal7z\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\akhav\.platformio\.cache\tmp-2199648I3sw0Wal7z\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\akhav\.platformio\.cache\tmp-2199648I3sw0Wal7z\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\akhav\.platformio\.cache\tmp-2199648I3sw0Wal7z\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\akhav\.plat...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-12-01:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

akhavea, 2017-12-02:
Got it thanks a ton

closed 2017-12-15 by ivankravets