platformio/platformio-atom-ide #1199: Uncaught Error: EPERM: operation not permitted, lstat '~/Documents\PlatformIO\Projects\ESP_OTA\src'

Gustavoprojetos, 2017-11-25:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.22.1 x64
**Electron**: 1.6.15
**OS**: Unknown Windows version
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: EPERM: operation not permitted, lstat 'C:\Users\Gustavo-pc\Documents\PlatformIO\Projects\ESP_OTA\src'

```
At fs.js:902

Error: EPERM: operation not permitted, lstat 'C:\Users\Gustavo-pc\Documents\PlatformIO\Projects\ESP_OTA\src'
    at fs.lstatSync (fs.js:902:18)
    at Object.fs.lstatSync (ELECTRON_ASAR.js:244:16)
    at Proxy.realpathSync (fs.js:1542:21)
    at Proxy.fs.realpathSync (ELECTRON_ASAR.js:321:29)
    at /packages/platformio-ide/lib/project/helpers.js:32:25
    at getActivePioProject (/packages/platformio-ide/lib/project/helpers.js:36:6)
    at Object.highlightActiveProject (/packages/platformio-ide/lib/maintenance.js:133:13)
    at doHighlight (/packages/platformio-ide/lib/main.js:227:43)
    at Function.module.exports.Emitter.simpleDispatch (~/AppData/Local/atom/app-1.22.1/resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (~/AppData/Local/atom/app-1.22.1/resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at ~/AppData/Local/atom/app-1.22.1/resources/app/src/workspace.js:550:30
```

### Commands

```
```

### Non-Core Packages

```
atom-beautify 0.30.9 
atom-ide-ui 0.6.0 
autocomplete-clang 0.11.4 
build 0.69.0 
busy 0.7.0 
busy-signal 1.4.3 
file-icons 2.1.14 
git-plus 7.10.0 
highlight-selected 0.13.1 
ide-csharp 0.6.1 
ide-flowtype 0.17.4 
ide-java 0.6.8 
ide-php 0.6.10 
ide-typescript 0.7.0 
intentions 1.1.5 
language-ini 1.19.0 
language-rust 0.4.12 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
minimap 4.29.7 
minimap-highlight-selected 4.6.1 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
tool-bar-main 0.0.10 
```

Gustavoprojetos, 2017-11-25:
j

ivankravets, 2017-12-01:
Something is wrong with permissions. Try to create new project and move source code from old.

closed 2017-12-01 by ivankravets