platformio/platformio-atom-ide #1204: C/C++ project index rebuild failed

azhornyak, 2017-11-26:
Error: [Sun Nov 26 17:16:55 2017] Processing lolin32 (platform: espressif32; board: lolin32; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PlatformioException: Could not load broken JSON: /Users/andrii/Documents/PlatformIO/Projects/171119-192551-lolin32/lib/ESP8266wifi/library.json:
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 156:
env.SConscript("$BUILD_SCRIPT")
File "/Users/andrii/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/andrii/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/Users/andrii/.platformio/platforms/espressif32/builder/main.py", line 163:
target_elf = env.BuildProgram()
File "/Users/andrii/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 64:
deplibs = env.BuildProjectLibraries()
File "/Users/andrii/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 676:
lib_builders = env.GetLibBuilders()
File "/Users/andrii/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 648:
env, join(libs_dir, item), verbose=verbose)
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 50:
env, path, verbose=verbose)
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 95:
self._manifest = manifest if manifest else self.load_manifest()
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piolib.py", line 485:
manifest = util.load_json(join(self.path, "library.json"))
File "/Users/andrii/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 192:
"Could not load broken JSON: %s" % file_path)
========================== [ERROR] Took 4.36 seconds ==========================

ivankravets, 2017-12-01:
Where did you get this library `/Users/andrii/Documents/PlatformIO/Projects/171119-192551-lolin32/lib/ESP8266wifi/`?

closed 2017-12-15 by ivankravets

ivankravets, 2018-01-10:
Known issue, see PR https://github.com/ekstrand/ESP8266wifi/pull/48/files

Please don't use RAW files and use PlatformIO Library Manager and remove that broken library. Add this line to your `platformio.ini` file
```
[env:..]
platform = ...
lib_deps = SerialESP8266wifi
```