platformio/platformio-atom-ide #1205: Completely non-functional

Sod-Almighty, 2017-11-26:
I've installed the PlatformIO package in Atom 1.22.1, and the entire PlatformIO menu is non-functional. None of the menu items work.

I've tried closing Atom and re-loading it. Doesn't help.

ivankravets, 2017-12-01:
What is your OS?

Sod-Almighty, 2017-12-01:
OSX El Crapitan 10.11.6

ivankravets, 2017-12-01:
Does your project contain symlink?

Sod-Almighty, 2017-12-01:
My project doesn't contain anything whatsoever, yet. I don't even have a project. I was trying to get the PlatformIO Home window to show up, so that I could *create* a project.

Sod-Almighty, 2017-12-01:
But in the fullness of time, yes, my projects *will* contain directory symlinks.

ivankravets, 2017-12-01:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

Sod-Almighty, 2017-12-01:
```
└───% echo $PATH
/usr/local/opt/ruby@2.3/bin:/usr/local/bin:/opt/sdcc/bin:/opt/bin:/usr/bin:/bin:/usr/sbin:/sbin

└───% echo $PYTHONPATH

└───% python --version
Python 2.7.10

└───% pip --version
zsh: command not found: pip
```
Presumably I should install pip? I don't recall seeing that in the instructions.

Sod-Almighty, 2017-12-01:
I've installed pip, but it doesn't seem to have helped.

```
└───% pip --version
pip 9.0.1 from /Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg (python 2.7)

└───% pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
altgraph (0.10.2)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
macholib (1.5.1)
matplotlib (1.3.1)
modulegraph (0.10.4)
numpy (1.8.0rc1)
pip (9.0.1)
py2app (0.7.3)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
python-dateutil (1.5)
pytz (2013.7)
scipy (0.13.0b1)
setuptools (1.1.6)
six (1.4.1)
vboxapi (1.0)
xattr (0.6.4)
zope.interface (4.1.1)
```

ivankravets, 2017-12-01:
Do you see any errors in Atom > View > Developer > Toggle Developer Tools > Console?

Please try to uninstall Atom and install again. 

- http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

Sod-Almighty, 2017-12-01:
I see plenty of errors, but none that seem related to this issue:
```
atom-parinfer v1.18.0 activated
<embedded>:67741 Invalid keystroke sequence for binding: `^-shift-up: multi-cursor-plus:select-up` in /Users/infinity/.atom/keymap.cson
module.exports.KeymapManager.build @ <embedded>:67741
<embedded>:67741 Invalid keystroke sequence for binding: `^-shift-down: multi-cursor-plus:select-down` in /Users/infinity/.atom/keymap.cson
module.exports.KeymapManager.build @ <embedded>:67741
/Users/infinity/.atom/packages/sync-settings/lib/sync-settings.coffee:93 checking latest backup...
/Users/infinity/.atom/packages/sync-settings/lib/sync-settings.coffee:392 Creating GitHubApi client with token = b1e327fb4a8cc523b68ba9401db995c581870bf0
/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:179 Uncaught (in promise) TypeError: tabBarView.addEventListener is not a function
    at Object.tabBarViewAdded (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:112:16)
    at Object.connectPaneWithTabBarView (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:80:10)
    at Object.paneAdded (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:51:10)
    at /Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:36:65
    at PaneContainer.observePanes (/Applications/Atom.app/Contents/Resources/app/src/pane-container.js:77:47)
    at WorkspaceCenter.observePanes (/Applications/Atom.app/Contents/Resources/app/src/workspace-center.js:162:37)
    at /Applications/Atom.app/Contents/Resources/app/src/workspace.js:908:32
    at Array.map (native)
    at Workspace.observePanes (/Applications/Atom.app/Contents/Resources/app/src/workspace.js:907:116)
    at Object.tabsActivated (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:36:41)
    at Object.packageActivated (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:28:8)
    at /Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:10:67
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at pack.activate.then (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:694:28)
/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:179 Uncaught (in promise) TypeError: tabBarView.addEventListener is not a function
    at Object.tabBarViewAdded (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:112:16)
    at Object.connectPaneWithTabBarView (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:80:10)
    at Object.paneAdded (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:51:10)
    at /Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:36:65
    at PaneContainer.observePanes (/Applications/Atom.app/Contents/Resources/app/src/pane-container.js:77:47)
    at WorkspaceCenter.observePanes (/Applications/Atom.app/Contents/Resources/app/src/workspace-center.js:162:37)
    at /Applications/Atom.app/Contents/Resources/app/src/workspace.js:908:32
    at Array.map (native)
    at Workspace.observePanes (/Applications/Atom.app/Contents/Resources/app/src/workspace.js:907:116)
    at Object.tabsActivated (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:36:41)
    at Object.packageActivated (/Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:28:8)
    at /Users/infinity/.atom/packages/tabasco/lib/tabasco.coffee:10:67
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at pack.activate.then (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:694:28)
/Users/infinity/.atom/packages/sync-settings/lib/sync-settings.coffee:118 latest backup version e3512676f6bfb9824f15d3c06032f5be2856e78f
events.js:101 Uncaught (in promise) Error: Fetching outdated packages and themes failed.
    at /Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:267:27
    at exit (/Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:128:22)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:338:17)
    at /Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:351:17
    at Socket.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:220:15)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:501:12)
```

If I uninstalled and reinstalled Atom, wouldn't I lose all my extensions and settings? That would be highly inconvenient.

ivankravets, 2017-12-01:
It seems that one of your custom package breaks Atom's Toolbar package. Does PIO Home work from `Menu: PlatformIO > Home`?

Sod-Almighty, 2017-12-01:
Selecting `PlatformIO > Home` does precisely nothing. None of the other menu items do anything either.

Sod-Almighty, 2017-12-15:
Nope

ivankravets, 2017-12-15:
Could you try to backup/rename `~/.atom` folder, and restart IDE? It seems that some of your extra extension conflicts.

Sod-Almighty, 2017-12-15:
I have renamed the directory, and restarted Atom, and of course there are no extensions. So I've installed PlatformIO and it went through a long-winded install process which it didn't do before. It now seems to be working.

However, I do kinda *use* my various extensions, so this isn't really going to solve the problem. If you think one of them is conflicting, then perhaps you could tell me which one?

Sod-Almighty, 2017-12-15:
While we're on the subject, can you explain why it is *completely impossible* to select text with the mouse from either the PlatformIO build output window, or the PlatformIO "list serial ports" window?

ivankravets, 2017-12-15:
This is known UI issue. We recommend switching to PIO IDE for VSCode. It has better integration.

Sod-Almighty, 2017-12-15:
Sigh.

Sod-Almighty, 2017-12-15:
You know, if you'd just admitted from the start that the damn extension was broken and unusable, I wouldn't've wasted my valuable time trying to get it working.

ivankravets, 2017-12-15:
You are the first with this issue for the last 2 years. Please provide a full list of installed Atom extensions, I'll check it later.

Sod-Almighty, 2017-12-15:
If the UI is broken anyway, what's the point? Without the ability to select and copy text, it cannot be used.

Besides, the only way I know of to give you a list of extensions is to type them in manually. And I'm not doing that.

ivankravets, 2017-12-15:
We support 10+ different IDE/Text Editors and can't fix their issues. We provide integration extensions for different IDE/Text Editors where users decide which to use.

We are responsible for the PlatformIO Core and its dev/platforms. 

Full list here http://docs.platformio.org/en/latest/ide.html

Sod-Almighty, 2017-12-16:
If you make an extension, you are responsible for fixing that extension. None of my other extensions go all bolloxed when I try to select text with the mouse. Only yours. Are you seriously trying to blame Atom for this?

zvonka, 2021-01-07:
Are you seriously expecting any kind of meaningful assistance with this kind of tone/attitude? 

> If you make an extension, you are responsible for fixing that extension. None of my other extensions go all bolloxed when I try to select text with the mouse. Only yours. Are you seriously trying to blame Atom for this?

infinity-computers-dot-net, 2021-01-07:
You've already admitted that it's broken, and that you're unwilling to fix it. So no, I'm not expecting any kind of assistance. 

> This is known UI issue. We recommend switching to PIO IDE for VSCode.

What I _would_ expect from any decent devs, is that they wouldn't leave KNOWN BROKEN SHIT up on the hub where people can waste their valuable time trying to install and use it.

Sod-Almighty, 2021-01-07:
You've already admitted that it's broken, and that you're unwilling to fix it. So no, I'm not expecting any kind of assistance.

This is known UI issue. We recommend switching to PIO IDE for VSCode.

What I would expect from any decent devs, is that they wouldn't leave KNOWN BROKEN SHIT up on the hub where people can waste their valuable time trying to install and use it.