platformio/platformio-atom-ide #1216: InstallationManager

terryb58, 2017-11-29:
Error: Error: Error: Failed to download file http://dl.platformio.org/ide-data/platformio-examples.tar.gz: 
    at ProjectExamplesStage.<anonymous> (file:///C:/Users/terryb.KLO/.atom/packages/platformio-ide/lib/installer/stages/project-examples.js:158:13)
    at Generator.throw (<anonymous>)
    at step (C:\Users\terryb.KLO\.atom\packages\platformio-ide\lib\installer\stages\project-examples.js:13:273)

ivankravets, 2017-12-01:
Where did you find this old installer? There is 2.0.0-rc.4 where this part of the code was removed.

closed 2017-12-01 by ivankravets

terryb58, 2017-12-01:
Hi Ivan,

I don't know.  It isn't a new download.  I was installing Atom and didn't
realize I had already installed it on this PC previously.  Then that
message popped up.  I removed platformio and reinstalled it and it seems to
be working fine now.

Terry