platformio/platformio-atom-ide #1223: Uncaught Error: Could not start PIO Home server: Traceback (most recent call last):  File "main...

betschart, 2017-12-03:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.22.1 x64
**Electron**: 1.6.15
**OS**: DESKTOP-TA1GPJP
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 21, in init cmd_debug
  File "C:\Users\pb\.platformio\packages\pysite-pioplus\twisted\internet\stdio.py", line 30, in <module>
    from twisted.internet import _win32stdio
  File "C:\Users\pb\.platformio\packages\pysite-pioplus\twisted\internet\_win32stdio.py", line 9, in <module>
    import win32api
ImportError: DLL load failed: %1 ist keine zul�ssige Win32-Anwendung.

```
At C:\Users\pb\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1571

Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 21, in init cmd_debug
  File "C:\Users\pb\.platformio\packages\pysite-pioplus\twisted\internet\stdio.py", line 30, in <module>
    from twisted.internet import _win32stdio
  File "C:\Users\pb\.platformio\packages\pysite-pioplus\twisted\internet\_win32stdio.py", line 9, in <module>
    import win32api
ImportError: DLL load failed: %1 ist keine zul�ssige Win32-Anwendung.

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at ChildProcess.cp.emit (/packages/platformio-ide/node_modules/cross-spawn/lib/enoent.js:40:29)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
     -2:42.4.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.69.0 
busy-signal 1.4.3 
file-icons 2.1.14 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

ivankravets, 2017-12-03:
Try to uninstall all existing Pythons from system and restart IDE

ivankravets, 2017-12-11:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

cyberjacob, 2017-12-11:
```
Microsoft Windows [Version 6.3.9600]
(c) 2013 Microsoft Corporation. All rights reserved.

C:\Users\Jacob>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\Ph
ysX\Common;C:\Users\Jacob\Documents\AMD APP\bin\x86_64;C:\Users\Jacob\Documents\
AMD APP\bin\x86;C:\Program Files (x86)\AMD APP\bin\x86_64;C:\Program Files (x86)
\AMD APP\bin\x86;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Wind
ows\System32\WindowsPowerShell\v1.0\;C:\MinGW\bin;C:\Program Files (x86)\ATI Tec
hnologies\ATI.ACE\Core-Static;C:\Program Files (x86)\Windows Live\Shared;C:\Prog
ram Files (x86)\Common Files\Acronis\SnapAPI\;C:\Program Files\Microsoft SQL Ser
ver\110\Tools\Binn\;C:\Program Files (x86)\Calibre2\;C:\Program Files (x86)\Micr
osoft SDKs\TypeScript\1.0\;C:\WINDOWS\System32\Windows System Resource Manager\b
in;C:\Program Files (x86)\GNU\GnuPG\pub;C:\Program Files (x86)\AMD\ATI.ACE\Core-
Static;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System
32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\
LLVM\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;c:\Program Files
\FileBot\;C:\Users\Jacob\AppData\Roaming\Dashlane\4.6.5.21982\bin\Firefox_Extens
ion\{442718d9-475e-452a-b3e1-fb1ee16b8e9f}\components;C:\Users\Jacob\AppData\Roa
ming\Dashlane\4.6.6.23032\bin\Firefox_Extension\{442718d9-475e-452a-b3e1-fb1ee16
b8e9f}\components;C:\Program Files (x86)\Nmap;C:\Users\Jacob\AppData\Roaming\Das
hlane\4.7.1.28303\bin\Firefox_Extension\{442718d9-475e-452a-b3e1-fb1ee16b8e9f}\c
omponents;C:\Users\Jacob\AppData\Roaming\Dashlane\4.8.9.37106\bin\Firefox_Extens
ion\{442718d9-475e-452a-b3e1-fb1ee16b8e9f}\components;C:\Users\Jacob\AppData\Roa
ming\Dashlane\4.8.9.37106\ucrt;C:\Users\Jacob\AppData\Roaming\Dashlane\5.2.0.116
59\bin\Firefox_Extension\{442718d9-475e-452a-b3e1-fb1ee16b8e9f}\components;C:\Us
ers\Jacob\AppData\Roaming\Dashlane\5.2.0.11659\ucrt;C:\Users\Jacob\AppData\Local
\atom\bin

C:\Users\Jacob>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\Jacob>python --version
'python' is not recognized as an internal or external command, operable program or batch file.

C:\Users\Jacob>pip --version
'pip' is not recognized as an internal or external command, operable program or batch file.

C:\Users\Jacob>pip list
'pip' is not recognized as an internal or external command, operable program or batch file.

C:\Users\Jacob>

```

ivankravets, 2017-12-12:
Please remove `%HOME_DIR/.platformio` folder and restart IDE.

cyberjacob, 2017-12-12:
No change, same error message :(

ivankravets, 2017-12-14:
OK,

1. Please navigate to System Control Center, remove all Python installations
2. Remove `%HOME_DIR/.platformio` folder
3. Restart IDE

Does it work now?

ivankravets, 2017-12-15:
This is a known issue.

1. Please navigate to OS Control Panel > Applications and uninstall ALL Pythons from a system
2. Remove `%HOME_DIR%/.platformio`
3. Start IDE.

Now all should work.

bangtiray, 2017-12-22:
thanks for solution.. 
im getting succes for the trouble.. 
thank you very much

closed 2017-12-22 by ivankravets

MOAMIndustries, 2017-12-28:
Same error in win10 with PlatformIO Home not loading.
Remove %HOME_DIR/.platformio/packages  was ineffective
Closed atom. Ended Python application through task manager
Restarted Atom and PlatformIO Home loaded with no error

Python v2.7.11
Atom 1.23.1 ia32