platformio/platformio-atom-ide #1227: Uncaught Error: Could not start PIO Home server: Error: Could not install 'tool-pioplus' with ver...

wongchunhung, 2017-12-04:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.22.1 x64
**Electron**: 1.6.15
**OS**: Ubuntu 16.04.3
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: Error: Could not install 'tool-pioplus' with version requirements '>=0.11.0,<2' for your system 'linux_x86_64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

```
At /home/ivan/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: Error: Could not install 'tool-pioplus' with version requirements '>=0.11.0,<2' for your system 'linux_x86_64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.69.0 
busy-signal 1.4.3 
file-icons 2.1.14 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

ivankravets, 2017-12-04:
Could you check internet connection and restart IDE?

s0170071, 2017-12-15:
same problem here. With and  without internet connection ;-)

ivankravets, 2017-12-15:
@s0170071 please open PIO IDE Terminal and type `pio home`. Do you see any errors?

ivankravets, 2017-12-15:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1171

closed 2017-12-15 by ivankravets