platformio/platformio-atom-ide #1228: C/C++ project index rebuild failed

OPatrick, 2017-12-04:
Traceback (most recent call last):
  File "C:\Users\Patrick\.platformio\penv\Scripts\platformio-script.py", line 11, in <module>
    load_entry_point('platformio==3.5.0rc5', 'console_scripts', 'platformio')()
  File "c:\users\patrick\.platformio\penv\lib\site-packages\platformio\__main__.py", line 123, in main
    maintenance.on_platformio_exception(e)
  File "c:\users\patrick\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 63, in on_platformio_exception
    telemetry.on_exception(e)
  File "c:\users\patrick\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 323, in on_exception
    skip_conditions.append("[API] Account: " in str(e))
  File "c:\users\patrick\.platformio\penv\lib\site-packages\platformio\exception.py", line 23, in __str__
    return Exception.__str__(self)
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufffd' in position 357: ordinal not in range(128)

ivankravets, 2017-12-04:
Do you have non-ascii chars in project path or inside `platformio.ini`?

OPatrick, 2017-12-04:
Hello,

Thanks to respond

My platformio.ini files are as such:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:heltec_wifi_lora_32]
platform = espressif32
board = heltec_wifi_lora_32
framework = arduino
lib_extra_dirs = C:\Users\Patrick\Documents\Arduino

;[env:esp32dev]
;platform = espressif32
;board = esp32dev
;framework = arduino
;lib_extra_dirs = C:\Users\Patrick\Documents\Arduino

My projects folders and lib_extra folders is the same 
(C:\Users\Patrick\Documents\Arduino), since I'm trying to build reusable 
libraries. Could that be the reason?

No non-ascii chars in project names...

ivankravets, 2017-12-04:
Could you reproduce this issue again after IDE restart?

OPatrick, 2017-12-04:
Well, after forced updates in the right order: atom then platformio then plugins, and reboot, all went OK
Do not know what happened.
Thanks for your help

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets