platformio/platformio-atom-ide #1231: C/C++ project index rebuild failed

XDedX, 2017-12-04:
ImportError: cannot import name parse_http_list:
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/main.py", line 26:
from platformio import util
File "/usr/local/lib/python2.7/dist-packages/platformio/util.py", line 34:
import requests
File "/usr/local/lib/python2.7/dist-packages/requests/__init__.py", line 97:
from . import utils
File "/usr/local/lib/python2.7/dist-packages/requests/utils.py", line 26:
from ._internal_utils import to_native_string
File "/usr/local/lib/python2.7/dist-packages/requests/_internal_utils.py", line 11:
from .compat import is_py2, builtin_str, str
File "/usr/local/lib/python2.7/dist-packages/requests/compat.py", line 42:
from urllib2 import parse_http_list
========================== [ERROR] Took 3.36 seconds ==========================

ivankravets, 2017-12-15:
Please run `pio update`. Could you reproduce this issue again?

ivankravets, 2018-01-10:
Please reopen if you still need help

closed 2018-01-10 by ivankravets

leisther, 2019-01-27:
how to solve this issue? badly need help :(

leisther, 2019-01-27:
Traceback (most recent call last):
  File "train.py", line 15, in <module>
    from debugger import Debugger
  File "/home/leisther/Downloads/deep-learning-for-human-part-discovery-in-images-master/debugger.py", line 4, in <module>
    import matplotlib
  File "/home/leisther/.local/lib/python3.6/site-packages/matplotlib/__init__.py", line 136, in <module>
    import urllib.request
  File "/usr/local/lib/python3.6/urllib/request.py", line 102, in <module>
    import requests
  File "/home/leisther/.local/lib/python3.6/site-packages/requests/__init__.py", line 112, in <module>
    from . import utils
  File "/home/leisther/.local/lib/python3.6/site-packages/requests/utils.py", line 26, in <module>
    from ._internal_utils import to_native_string
  File "/home/leisther/.local/lib/python3.6/site-packages/requests/_internal_utils.py", line 11, in <module>
    from .compat import is_py2, builtin_str, str
  File "/home/leisther/.local/lib/python3.6/site-packages/requests/compat.py", line 58, in <module>
    from urllib.request import parse_http_list, getproxies, proxy_bypass, proxy_bypass_environment, getproxies_environment
ImportError: cannot import name 'parse_http_list'