platformio/platformio-atom-ide #1233: file open dialog is not displayed

ja1krg, 2017-12-05:
[image]

The file open dialog does not work.
When trying to specify Location from New Project, when opening Import Arduino Project, when opening Open Project, it stops in the state of attached image.
Installing the OS newly on the PC, installing Atom in the required update, and then installing PlatformIO.
If you have something to know, please let me know.

The environment is as follows.
Windows 10 Pro 64bit
Atom 1.22.1 x 64
PlatformIO Home 0.5.0 / Core 3.5.0 rc 5

ivankravets, 2017-12-15:
Please open PIO IDE Terminal and type `pio home`. Do you see any errors when selecting Open Project?

ivankravets, 2018-01-10:
Please open PIO IDE Terminal and run
```
pio upgrade
pio update
```

It should work now.

closed 2018-01-10 by ivankravets