platformio/platformio-atom-ide #1238: InstallationManager

kurniawan77, 2017-12-05:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Irwan\.platformio\.cache\tmp-1002803NtiOXzfOzu\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Irwan\.platformio\.cache\tmp-1002803NtiOXzfOzu\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Irwan\.platformio\.cache\tmp-1002803NtiOXzfOzu\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Irwan\.platformio\.cache\tmp-1002803NtiOXzfOzu\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\Irwan\.platformio\.cache\tmp-1002803NtiOXzfOzu\virtualenv-14.0.6\virtualenv.py", line 743, in call_subprocess
    cwd=cwd, env=env)
  File "C:\Users\Irwan\.platformio\python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Users\Irwan\.platformio\python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

ivankravets, 2018-01-10:
@mosescomp @kurniawan77 @DocSlump @kochalois
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

kurniawan77, 2018-01-10:
I reinstalled everything... got it working again

ivankravets, 2018-01-10:
@mosescomp @DocSlump @kochalois @Jakim121
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

Also, try to disable antivirus for a while and restart IDE.

Jakim121, 2018-01-10:
I disabled antivirus for a while, it is fixed.

mosescomp, 2018-01-11:
I let PlatformIO send the error report to you.  I believe that it was a ‘hiccup’. 

Now that I have been using it a while and have gotten ALL of the updates, it appears to be working flawlessly.

Thank you for creating an absolutely fantastic product.  It is a pleasure to use, and by the way, super fast!

On the MacBook I couldn’t get scroll back to work in the terminal window - until I connected a mouse, fine now!  Hmm, maybe I would have encountered the same problem on the PC also, if I hadn’t had a mouse connected to it.

Thank you!
Mike Moses

ivankravets, 2018-01-11:
I added all solutions for this issue to our FAQ: http://docs.platformio.org/en/latest/faq.html#package-manager

ivankravets, 2018-06-20:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1171

closed 2018-06-20 by ivankravets