platformio/platformio-atom-ide #1240: InstallationManager

brainey, 2017-12-05:
Occurred after starting up atom and installation manager started to do it's upgrade I'm assuming ...
Had 9 project windows open - 7 failed and it looks like two succeeded with platformIO ide install....

Virtualenv Create: Traceback (most recent call last):
  File "/home/brainey/.platformio/.cache/tmp-3150WviNAmNPR1Bv/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/home/brainey/.platformio/.cache/tmp-3150WviNAmNPR1Bv/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/home/brainey/.platformio/.cache/tmp-3150WviNAmNPR1Bv/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/brainey/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath
    if not isabs(path):
  File "/home/brainey/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2018-01-30:
@BahmanK , @mdanetzky, @brainey

Have you resolved this issue?

ivankravets, 2018-06-20:
Please upgrade `platformio-ide` to 2.1.5

closed 2018-06-20 by ivankravets