platformio/platformio-atom-ide #1243: InstallationManager

PierreScerri, 2017-12-06:
This error comes up halfway through installation of platform-ide in Atom:
PIP: Error: spawn /Volumes/Macintosh HD/users/pierrescerri/.platformio/penv/bin/pip ENOENT
What should I do?

I'm on macOS high Sierra.

Thank you.

ivankravets, 2017-12-15:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1232

PierreScerri, 2017-12-15:
Hi

Atom wants to restart every time now (Green alert box)
And I'm now getting this error:

Error: spawn platformio ENOENT

Terminal output:
Pierres-iMac:~ pierrescerri$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/opt/X11/bin
Pierres-iMac:~ pierrescerri$ $PYTHONPATH
Pierres-iMac:~ pierrescerri$ python --version
Python 2.7.10
Pierres-iMac:~ pierrescerri$ pip --version
-bash: pip: command not found
Pierres-iMac:~ pierrescerri$ pip list
-bash: pip: command not found
Pierres-iMac:~ pierrescerri$ 

Thank you.

Pierre Scerri
pierre.scerri@gmail.com

AeroGirl, 2017-12-18:
Hi,

I am getting the exact same error when attempting to install platform-ide.

**OS:** MacOS
**Atom:** 1.23.0
**platform-ide:** Current aval through package manager in Atom

**echo $PATH**
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin

**echo $PYTHONPATH**
[NOTHING RETURNED]

**python --version**
2.7.10

**pip --version**
9.0.1 from /Library/Python/2.7/site-packages/pip-0.0.1-py2.7.egg (python 2.7)

**pip list**
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
altgraph (0.10.2)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
macholib (1.5.1)
matplotlib (1.3.1)
modulegraph (0.10.4)
numpy (1.8.0rc1)
pip (9.0.1)
py2app (0.7.3)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
python-dateutil (1.5)
pytz (2013.7)
scipy (0.13.0b1)
setuptools (18.5)
six (1.4.1)
xattr (0.6.4)
zope.interface (4.1.1)

[image]

ivankravets, 2017-12-18:
@AeroGirl does this file exist `/Volumes/DATA SSD/Users/meimcounting/.platformio/penv/bin/pip`?

ivankravets, 2017-12-27:
@werlley007 , @vitaltr

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2017-12-27:
@werlley007 , @vitaltr, @alexfederlin

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2017-12-30:
@werlley007 , @vitaltr, @alexfederlin, @thehellmaker, @bruster999

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2018-01-06:
@werlley007 , @vitaltr, @alexfederlin, @thehellmaker, @bruster999, @Egaro555, @Rukyr97

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2018-01-06:
@hygorjardim

Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

Rukyr97, 2018-01-06:
[image]

hygorjardim, 2018-01-06:
[image: Imagem inline 2]

I'm sorry, maybe I did not pay attention to this error, I did not try to
fix it right now. I sent the error thinking of helping the developers of
the platform.

Rukyr97, 2018-01-06:
i've been trying to reinstall platformio packages and atom several times... restarted the pc... Nothing seems to be helping.

ivankravets, 2018-01-10:
@werlley007 , @vitaltr, @alexfederlin, @thehellmaker, @bruster999, @Egaro555, @Rukyr97, @papa-ping

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2018-01-10:
Could someone provide a TeamViewer session to debug this issue? I would be thankful! 

You can reach me via me@ikravets.com.

Rukyr97, 2018-01-10:
i have posted the system output of the commands u told me to use in cmd... maybe you didn't see it? Last time i used Teamviewer i got some problems, so i prefer avoiding it... if it is possible we could use other software that uses screensharing?

You can reach me via me@ikravets.com<mailto:me@ikravets.com>.

ivankravets, 2018-01-10:
@Rukyr97 Do you know alternatives to TeamViewer? I think you can remove it after our debug session. You will have all control and see what I will do. Thanks!

Rukyr97, 2018-01-10:
Ok, i will install it

alexfederlin, 2018-01-11:
1. removed ~/.platformio/penv
2. started atom
3. Got green message that atom restart was required two consecutive times
4. on the 3rd time I got 
```
C/C++ project index rebuild failed
Error: spawn platformio ENOENT
```

Output from Systemterminal
```
alex@gina:~/jukebox/Provisioning/expresspopulator$ cd
alex@gina:~$ echo $PATH
/home/alex/perl5/bin:/home/alex/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/alex/bin:/opt/linuxtrack/bin
alex@gina:~$ echo $PYTHONPATH
/usr/local/lib/python2.7/dist-packages
alex@gina:~$ python --version
Python 2.7.12
alex@gina:~$ pip --version
pip 8.1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
alex@gina:~$ pip list
adium-theme-ubuntu (0.3.4)
alabaster (0.7.7)
ansible (2.4.2.0)
apsw (3.8.11.1.post1)
arrow (0.12.0)
attrs (15.2.0)
awscli (1.14.14)
Babel (1.3)
backports.functools-lru-cache (1.2.1)
BeautifulSoup (3.2.1)
beautifulsoup4 (4.4.1)
blinker (1.3)
botocore (1.8.18)
bottle (0.12.13)
ccsm (0.9.12.2)
certifi (2017.11.5)
chardet (2.3.0)
Cheetah (2.4.4)
CherryPy (3.5.0)
click (5.1)
colorama (0.3.9)
command-not-found (0.3)
compizconfig-python (0.9.12.2)
cryptography (1.2.3)
cssselect (0.9.1)
cssutils (1.0)
cycler (0.9.0)
debtags (2.0)
decorator (4.0.6)
defer (1.0.6)
deluge (1.3.12)
dirspec (13.10)
discid (1.1.0)
dnspython (1.12.0)
docutils (0.12)
duplicity (0.7.6)
ecdsa (0.13)
enum34 (1.1.2)
evernote (1.25.2)
eyeD3 (0.6.18)
feedparser (5.1.3)
futures (3.2.0)
geeknote (0.2a0)
gyp (0.1)
hachoir-core (1.3.3)
hachoir-metadata (1.3.3)
hachoir-parser (1.3.4)
html2text (2017.10.4)
html5lib (0.999)
httplib2 (0.9.1)
idna (2.0)
imaplib2 (2.53)
ipaddr (2.1.11)
ipaddress (1.0.16)
Jinja2 (2.8)
jmespath (0.9.3)
libvirt-python (1.3.1)
lockfile (0.12.2)
lxml (3.5.0)
Mako (1.0.3)
Markdown (2.6.6)
markdown2 (2.3.4)
MarkupSafe (0.23)
matplotlib (1.5.1)
mbed-cli (1.2.2)
mechanize (0.2.5)
meld (3.14.2)
musicbrainzngs (0.6)
mutagen (1.31)
ndg-httpsclient (0.4.0)
netifaces (0.10.4)
networkx (1.11)
numpy (1.11.0)
oauth2 (1.9.0.post1)
oauthlib (1.0.3)
oneconf (0.3.9)
PAM (0.4.2)
paramiko (1.16.0)
pexpect (4.0.1)
Pillow (3.1.2)
pip (8.1.1)
piston-mini-client (0.7.5)
plotly (1.9.5)
ptyprocess (0.5)
pyasn1 (0.1.9)
pyasn1-modules (0.0.7)
pycrypto (2.6.1)
pycups (1.9.73)
pycurl (7.43.0)
pygame (1.9.1release)
Pygments (2.1)
pygobject (3.20.0)
pygpgme (0.3)
PyJWT (1.3.0)
PyOpenGL (3.0.2)
pyOpenSSL (0.15.1)
pyparsing (2.0.3)
PyPDF2 (1.25.1)
pyserial (3.0.1)
pysmbc (1.0.15.5)
PySocks (1.5.0)
Pyste (0.9.10)
python-apt (1.1.0b1)
python-dateutil (2.4.2)
python-debian (0.1.27)
python-libdiscid (0.4.1)
python-libtorrent (1.0.7)
pytz (2014.10)
pyxdg (0.25)
PyYAML (3.11)
pyzmq (15.2.0)
reportlab (3.3.0)
repoze.lru (0.6)
requests (2.9.1)
roman (2.0.0)
Routes (2.2)
rsa (3.4.2)
s3transfer (0.1.12)
scipy (0.17.0)
scour (0.32)
semantic-version (2.6.0)
service-identity (16.0.0)
setuptools (20.7.0)
six (1.10.0)
software-center-aptd-plugins (0.0.0)
speedtest-cli (0.3.4)
Sphinx (1.3.6)
sphinx-rtd-theme (0.1.9)
SQLAlchemy (1.2.0b2)
thrift (0.10.0)
Twisted (16.0.0)
ubuntuone-client-data (14.4)
unity-lens-photos (1.0)
urllib3 (1.13.1)
urwid (1.3.1)
uTidylib (0.2)
vboxapi (1.0)
WebOb (1.5.1)
wheel (0.29.0)
wxPython (3.0.2.0)
wxPython-common (3.0.2.0)
zenmap (7.1)
zope.interface (4.1.3)
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
alex@gina:~$ 
```

ivankravets, 2018-01-11:
Could you provide https://www.teamviewer.com/en/ session to debug this issue? Please contact me@ikravets.com. I would be thankful!

ivankravets, 2018-01-11:
We have deployed new installer where fixed the issue with `conda` tool. If you use this tool, please UNINSTALL our extension from Atom and INSTALL again. It should use a new installer.

Please inform us if that helped you.

alexfederlin, 2018-01-12:
Ubuntu 16.04:
well, I deleted both .atom and .platformio folders (I guess that's as good as reinstalling Atom?) and now I'm in a loop. 
I open the fresh atom, go to the Package installer
I search for platformio and select to install platformio.ide (2.0.0.rc7)
I get the green box that tells me to restart atom, 
I click ok, 
atom starts back up, 
tells me it installs platformio, 
gives me the green box, ... and so on. It does not stop...

ivankravets, 2018-01-16:
The issue was linked with broken Python's `arrow` package. We rewrote this part of a code in PlatformIO Core and removed it. Please UNINSTALL and INSTALL this extension in Atom > Packages.

Does it work now?

ivankravets, 2018-03-05:
@technorider

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

technorider, 2018-03-06:
halo. please check your email. because i'm send to u teamviewer ID thanks

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

commit c919f546 (2018-01-10): Init project indexer after PIO Core installer // Resolve #1243
closed 2020-06-13 by ivankravets