platformio/platformio-atom-ide #1245: InstallationManager

matthew0217, 2017-12-08:
Virtualenv Create: Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp-56089wgtumMLwrLt\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\.platformio\.cache\tmp-56089wgtumMLwrLt\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\.platformio\.cache\tmp-56089wgtumMLwrLt\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\.platformio\.cache\tmp-56089wgtumMLwrLt\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\.platformio\.cache\tmp-56089wgtumMLwrLt\virtualenv-14.0.6\virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\.platformio\penv\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2017-12-19:
@ghost, @ehyunsu123 

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2018-01-10:
Can you reproduce this issue with the latest version of PlatformIO IDE 2.0.0-rc.6?

ivankravets, 2018-01-16:
@ghost, @ehyunsu123, @klren0312

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

klren0312, 2018-01-26:
```
PS C:\Users\zzes> echo %PATH%
%PATH%
PS C:\Users\zzes> echo %PYTHONPATH%
%PYTHONPATH%
PS C:\Users\zzes> python --version
Python 2.7.14
PS C:\Users\zzes> pip --version
pip 9.0.1 from d:\python27\lib\site-packages (python 2.7)
PS C:\Users\zzes> pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
gevent (1.2.2)
greenlet (0.4.12)
msgpack-python (0.4.8)
pip (9.0.1)
setuptools (28.8.0)
```

ivankravets, 2018-01-30:
@ghost, @ehyunsu123, @raulibi

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ivankravets, 2018-01-30:
@klren0312 please use CMD.EXE, not PS.

klren0312, 2018-01-30:
```
C:\Users\zzes>echo %PATH%
D:\Python27\;D:\Python27\Scripts;E:\gradle\gradle-4.1\bin;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;E:\Git\cmd;E:\java\jdk\bin;E:\java\jdk\jre\bin;E:\gort_hardware;E:\apache-maven-3.3.9\bin;E:\sqlite;D:\maibu\VisualMWatch_W64\mbcc\bin;E:\GNUforARM\bin;E:\qampp\Data_lib\perl\bin;E:\qampp\apache\bin;E:\qampp\php;E:\qampp\svnserver\Subversion\bin;E:\qampp\mysql\bin;C:\ProgramData\ComposerSetup\bin;E:\apache-ant-1.10.1\bin;E:\mongodb\bin;E:\PostgreSQL\pg96\bin;E:\play-2.2.4;E:\erl9.0\bin;E:\android-sdk\tools;E:\android-sdk\platform-tools;E:\Go\bin;E:\Ruff SDK\bin;E:\phantomjs-2.1.1-windows\bin;E:\Goproject\bin;C:\Program Files\dotnet\;E:\Mercurial;E:\LLVM\bin;C:\Program Files (x86)\ZeroTier\One\;E:\nodejs\;E:\RailsInstaller\Ruby2.2.0\bin;C:\Users\zzes\AppData\Local\Microsoft\WindowsApps;C:\Users\zzes\AppData\Roaming\Composer\vendor\bin;E:\Microsoft VS Code Insiders\bin;C:\Users\zzes\AppData\Roaming\npm

C:\Users\zzes>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\zzes>python --version
Python 2.7.14

C:\Users\zzes>pip --version
pip 9.0.1 from d:\python27\lib\site-packages (python 2.7)

C:\Users\zzes>pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
gevent (1.2.2)
greenlet (0.4.12)
msgpack-python (0.4.8)
pip (9.0.1)
setuptools (28.8.0)
```

ivankravets, 2018-01-30:
Could try to open system terminal and type `pip install virtualenv`. Restart IDE. Does it work now?

cosmikwolf, 2018-03-16:
this problem is still happening for me on 2.1.3, and I think it has to do with an inability to install virtualenv. 

I get this error:
SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)

I think THAT has to do with the SSL version on my system:
MacPro:~ tenkai$ openssl version
OpenSSL 0.9.8zh 14 Jan 2016

I am on Mac OS 10.11.6 (can't update at the moment)
I can't update OpenSSL very easily, have to reboot and go into recovery mode to turn of system protections first.. lets see how this goes.

cosmikwolf, 2018-03-16:
Ok, after following the instructions on updating openSSL here: https://medium.com/@katopz/how-to-upgrade-openssl-8d005554401 I was able to install and get virtualenv to work. I installed virtualenv  separately, so the atom installer itself may have worked, but i diddnt check before doing the installation.

klren0312, 2018-03-17:
#### I also can not go well
```
C:\Users\60574>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;E:\Python27\;E:\Python27\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;E:\nvm;E:\nodejs;E:\android-sdk\platform-tools;E:\android-sdk\tools;E:\java8\jdk\bin;.;E:\java8\jdk\lib;E:\java8\jdklib\tool.jar;E:\Git\cmd;E:\qampp\Data_lib\perl\bin;E:\qampp\apache\bin;E:\qampp\php;E:\qampp\svnserver\Subversion\bin;E:\qampp\mysql\bin;E:\qampp\tomcat\bin;E:\qampp\postgresql\bin;E:\gradle-4.5.1\bin;E:\apache-maven-3.3.9\bin;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\WINDOWS\System32\OpenSSH\;E:\LLVM\bin;E:\python36\Scripts\;E:\python36\;C:\Users\60574\AppData\Local\Microsoft\WindowsApps;E:\nvm;E:\nodejs;E:\Microsoft VS Code Insiders\bin;E:\Egret Wing 3\bin

C:\Users\60574>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\60574>python --version
Python 2.7.14

C:\Users\60574>pip --version
pip 9.0.1 from e:\python27\lib\site-packages (python 2.7)

C:\Users\60574>pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
pip (9.0.1)
setuptools (28.8.0)
virtualenv (15.1.0)
```

#### there is my error log 
```
If you use Antivirus, it can block PlatformIO Installer. Try to disable it for a while.
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\60574\.platformio\.cache\tmp-7852SIl3PKN24koS\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\60574\.platformio\.cache\tmp-7852SIl3PKN24koS\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\60574\.platformio\.cache\tmp-7852SIl3PKN24koS\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\60574\.platformio\.cache\tmp-7852SIl3PKN24koS\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\60574\.platformio\.cache\tmp-7852SIl3PKN24koS\virtualenv-14.0.6\virtualenv.py", line 743, in call_subprocess
    cwd=cwd, env=env)
  File "E:\Python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "E:\Python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] 
Report a problem

```

ivankravets, 2018-06-20:
Please upgrade to `platformio-ide` 2.1.5, should be fixed. 
Please reopen if you still need help.

closed 2018-06-20 by ivankravets