platformio/platformio-atom-ide #1250: C/C++ project index rebuild failed

Pawlowski123, 2017-12-09:
Error: [Sat Dec  9 17:49:45 2017] Processing mega_pic32 (platform: microchippic32; board: mega_pic32; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting main.ino
sh: 1: pic32-g++: not found
*** Error 127
AssertionError: :
File "/home/marcelopawlowski/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/main.py", line 157:
env.SConscript("$BUILD_SCRIPT")
File "/home/marcelopawlowski/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/marcelopawlowski/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/marcelopawlowski/.platformio/platforms/microchippic32/builder/main.py", line 129:
target_elf = env.BuildProgram()
File "/home/marcelopawlowski/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/marcelopawlowski/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 60:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "/home/marcelopawlowski/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/marcelopawlowski/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 258:
env.ConvertInoToCpp()
File "/home/marcelopawlowski/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/marcelopawlowski/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "/home/marcelopawlowski/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 54:
return self.process(contents)
File "/home/marcelopawlowski/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
========================== [ERROR] Took 0.60 seconds ==========================

Pawlowski123, 2017-12-09:
I've tried to update PIO at the PlatformIO Terminal, all updated. Tried to download using the PlatformIO Home the frameworks and stuff. Tried to update gcc and g++ at Ubuntu's terminal, all goes well about updating, but always get the same error. pic32-gcc not found

ivankravets, 2017-12-15:
Does this binary work?
```
~/.platformio/packages/toolchain-microchippic32/bin/pic32-g++ --version
```

Pawlowski123, 2017-12-17:
When I try to run this binary it shows as "file or directory not found", when I run only g++ --version, it shows me

g++ (Ubuntu 5.4.0-6ubuntu1~16.04.5) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

G-Pereira, 2017-12-28:
any solution for this?

Pawlowski123, 2017-12-28:
Until now, nothing.

ivankravets, 2017-12-28:
Could you try [this](https://github.com/platformio/platformio-core/blob/develop/.travis.yml#L26)?

```
sudo apt-get install libc6-i386
```

closed 2018-06-21 by ivankravets