platformio/platformio-atom-ide #1253: Upload error in platformio

CaptainRon47, 2017-12-10:
I am working with Sarfata/kbox-firmware latest master.   I first did a pip update of platformio. I am using Atom to modify, compile and upload the files to the Kbox hardware.   I can compile and upload the host section of the program with no errors.  I can compile the 'esp' part of the program with no errors, but I get an error when I try to upload (esp).  Below is the output when I run 'platormio run -e esp -t upload' in a terminal (same error when I run it under the Atom environment.)  Thomas Sarfata has asked me to open a github (platformio/Atom) issue for this item.

thanks,
Ron Scotti

_[12/10/17 14:49:28] Processing esp (extra_scripts: tools/platformio_cfg_esp.py; framework: arduino; lib_deps: https://github.com/ttlappalainen/NMEA2000#b5426b3, https://github.com/bblanchon/ArduinoJson#126f7ab, Adafruit NeoPixel, elapsedMillis, https://github.com/me-no-dev/ESPAsyncTCP#a57560d, https://github.com/me-no-dev/ESPAsyncWebServer#63b5303; build_flags: -Wall -Isrc/common -Isrc/esp -DHAVE_STRLCPY -DHAVE_STRLCAT; lib_ignore: NMEA2000-Teensy, FlexCAN_Library, Time; platform: espressif8266; board: esp_wroom_02; lib_ldf_mode: deep; src_filter: +<common/*>,+<esp/*>)

Verbose mode can be enabled via `-v, --verbose` option
Collected 45 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <NMEA2000-library> v4.3.5
|   |-- <SPI> v1.0
|-- <ArduinoJson> v5.11.2
|-- <Adafruit NeoPixel> v1.1.3
|-- <elapsedMillis> v1.0.4
|-- <ESPAsyncTCP> v1.1.2
|   |-- <ESP8266WiFi> v1.0
|-- <ESP Async WebServer> v1.1.0
|   |-- <Hash> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <ArduinoJson> v5.11.2
|   |-- <ESPAsyncTCP> v1.1.2
|   |   |-- <ESP8266WiFi> v1.0
|-- <KBoxLogging>
|-- <ESP8266WiFi> v1.0
Looking for upload port...
Auto-detected: COM3
Uploading .pioenvs\esp\firmware.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...

A fatal error occurred: Invalid head of packet ('\xcd')

 [SUMMARY] 
Environment host       	[SKIP]
Environment program-esp	[SKIP]
Environment mfg        	[SKIP]
Environment test       	[SKIP]
Environment sktool     	[SKIP]_

ivankravets, 2017-12-15:
Could you try staging version? http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

CaptainRon47, 2017-12-15:
Ivan,

 

I have a question about the request to try the staging version.

Could you try staging version? http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

 

 

The ‘esp’ environment of the KBox project program BUILDS without errors, it is the UPLOAD to the Teensy where I get the error.   The instructions you sited only compare the BUILD of the project under the two IDE’s (PlatformIO and Arduino), do you want me to try the UPLOAD in both as well?

 

I did modify the platformio.ini file for the [env:esp] section of the project as below and that BUILT successfully, but it did so under the original platform statement (teensy) as well.

 

Thanks,

Ron

 

…

[env:esp]

src_filter = +<common/*>,+<esp/*>

# TODO: enable -Werror once platformio published v2.4 of esp8266/Arduino (pr #3271)

build_flags = -Wall -Isrc/common -Isrc/esp -DHAVE_STRLCPY -DHAVE_STRLCAT

platform = https://github.com/platformio/platform-espressif8266.git#feature/stage

framework = arduino

board = esp_wroom_02

extra_scripts = tools/platformio_cfg_esp.py

lib_ldf_mode = deep

lib_ignore = NMEA2000-Teensy, FlexCAN_Library, Time

lib_deps =

  ${common.lib_deps_common}

  Adafruit NeoPixel

  elapsedMillis

  https://github.com/me-no-dev/ESPAsyncTCP#a57560d

  https://github.com/me-no-dev/ESPAsyncWebServer#63b5303

…

ivankravets, 2017-12-15:
I don't see a problem with IDE. Please post your issue to https://community.platformio.org with all details.

closed 2017-12-15 by ivankravets

CaptainRon47, 2017-12-15:
Note: I can perform the upload succesfully using the command (using an esptool.py program)
**_esptool.py -p COM3 write_flash 0x0 .pioenvs\esp\firmware.bin_**
from a Windows Command window.
thanks

ivankravets, 2017-12-15:
Let's continue discussion here https://community.platformio.org/